wled/WLED #319: Yellow color on boot for 1 second

shantur, 2019-11-03:
Hi @Aircoookie 

Awesome project. Really like it.
I am trying to setup Wemos D1 Minis to control 300 pixels of SK6812 ( 5m x 60 per / m) strip. I have done the following

1. Erased D1 mini with `esptool.py --port /dev/tty.usbserial-1420 erase_flash`
2. Flashed `esptool.py --port /dev/tty.usbserial-1420 write_flash -fs 4MB -fm dout 0x0 ~/Downloads/wled_0.8.6_ESP8266_lwip1.bin`
3. Configured default color to 0(R) 0(G) 0(B) 117(W) in LED settings.
4. LED Strip and D1 mini are powered with 5V 5A brick.

Whenever power is removed and reapplied to both strip and D1 mini, the whole strip lights up as yellowish color first for 1 second before changing to correct (white) color.

Am I missing something or is there any way to fix this.

Thanks

Aircoookie, 2019-11-03:
Hi!

Awesome that you like it!
I too have noticed this issue, it should be fixed already in latest master source.
Here is a binary for you: https://drive.google.com/open?id=1AyVIWS6wnKRfQl2iSfFKXV7qJX4ewZf7
Let me know if that fixes the issue!

shantur, 2019-11-03:
Thanks for replying to my issue and the binary.
I tried this binary and now only the first pixel turns on as yellow at boot, then all turns off and around a second from power on all pixels are turned on with correct color.

This would work for me as most probably I will be hiding first pixel.

closed 2019-11-03 by shantur