wled/WLED #320: Flash issues esp32!!

LittleDevil76, 2019-11-03:
Getting  \No module named '_queue'/ while flashing esp32..!

Quick fix, or not.?

raoulteeuwen, 2019-11-03:
What version of Python are you using? Might be Python2 <> Python 3 related, where notation of queue was changed (Queue <> queue).

LittleDevil76, 2019-11-03:
I am just using a simple ESPHome flasher, and get this issue sometimes..

Aircoookie, 2019-11-04:
@LittleDevil76 I can't pinpoint a specific reason for why the flashing doesn't work, but here are a few points worth noting:
- Does your usb cable support data? If the ESP doesn't show up at all in your available serial ports, try a different cable.
- For some ESP models, you need to hold down the button labeled "boot" for the flash process to succeed

LittleDevil76, 2019-11-04:
Yes i need to hold down the Boot button on my boards, but this happens sometimes, not all the time.
Just a bit strange. 
And another problem i got with those boards is that dont hold the WIFI settings after power down and sometimes in mid use.
 No problems with the D1 mini bords tho.

lpacchin, 2019-11-09:
Hello
I have trouble with esp32, I can flash your already compiled bin but the board keep restarting with error, same if I compile with arduino 1.0.2 (as you suggested)
sudo python esptool.py --baud 921600 --port /dev/tty.usbserial-A5XK3RJT write_flash -fm dio -fs detect 0x10000 WLED_0.8.6_ESP32.bin 
Do you have any idea? WHat I'am doing wrong?
Thank you very much!

DirkTas67, 2019-11-09:
Hello,

I'm having problems too, I use a ESP32 D1 Mini (don't know if that's the exact name), tried several times to flash it (I use Ubuntu 18.04, esptool.py version 2.8), but that didn't work (didn't get any errors but never got to see the wled_ap), then I connected the esp32 mini (this thing does not have a boot button) directly to my home assistant machine, i could flash esphome without a problem and when enabling the webpage in the yaml I could upload the wled 0.8.6 esp32 pin 16 version from the webpage, however I never see a wled_ap.

Anybody a solution?  I'm now trying to make the binary myself but that can take some time since I don't have platformio installed.

Regards,
Dirk

raoulteeuwen, 2019-11-09:
> I have trouble with esp32, I can flash your already compiled bin but the board keep restarting with error, same if I compile with arduino 1.0.2 (as you suggested)
> sudo python esptool.py --baud 921600 --port /dev/tty.usbserial-A5XK3RJT write_flash -fm dio -fs detect 0x10000 WLED_0.8.6_ESP32.bin

Can you describe exact steps you take, at what moment the board restarts, your connections, your equipment, how you're powering the board, at what step you get an error, what error...

raoulteeuwen, 2019-11-09:
> I'm having problems too, I use a ESP32 D1 Mini (don't know if that's the exact name), tried several times to flash it (I use Ubuntu 18.04, esptool.py version 2.8), but that didn't work (didn't get [...]

Can you share the exact command you use to flash?

DirkTas67, 2019-11-09:
> Can you share the exact command you use to flash?

esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 /home/dirk/git/binary/wled/WLED_0.8.6_1911081_ESP32_ledpin16.bin

lpacchin, 2019-11-09:
@raoulteeuwen 
I'am using FTDI USB To TTL Serial Converter with esptool.py or arduino IDE
After flashing in the serial monitor I have this:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

python esptool.py --chip esp32 --baud 921600 --port /dev/tty.usbserial-A5XK3RJT write_flash -fm dio -fs detect 0x10000 WLED_0.8.6_ESP32.bin

umairyounus, 2019-11-10:
I'm having similar issues.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3f400020,len:265604

raoulteeuwen, 2019-11-10:
Sorry, no idea than, so i hope someone else replies...

lpacchin, 2019-11-10:
I have the same issue with esptool.py but when I compile and upload with arduino IDE works without problems, I personally found a little bit hard with all lib dependencies install in Arduino for the esp32, and it was successful only choosing the right board, maybe the already compiled bin works with only a type of esp32, mine are DOIT ESP32 DEV KIT V1 and did not work

DirkTas67, 2019-11-11:
> I have the same issue with esptool.py but when I compile and upload with arduino IDE works without problems, I personally found a little bit hard with all lib dependencies install in Arduino for the [...]

What kind do I need to get installed, can you tell me?

lpacchin, 2019-11-11:
@DirkTas67 
I try
You have to add the additional board url in arduino preferences:
https://dl.espressif.com/dl/package_esp32_index.json
Added the board in board manager Board manager and serch for esp32->install
(if you already have another url, please separate with comma for example I use also the esp8266 and is: 

[https://dl.espressif.com/dl/package_esp32_index.json, https://arduino.esp8266.com/stable/package_esp8266com_index.json]

then I added the following libraries that were required and gave error (not found) in compiling the sketch
1. SPI-master (https://github.com/PaulStoffregen/SPI)
2. azure-iot-arduino-master (https://github.com/Azure/azure-iot-arduino)
3. AsyncTCP-master (https://github.com/me-no-dev/AsyncTCP)
4. esp32 if do not have already (https://github.com/espressif/arduino-esp32)

After installing missing libraries the sketch compiled smootly and uploaded to the board
Check if your board definition is available and select before compiling in board manager

DirkTas67, 2019-11-11:
@lpacchin 

Arduino: 1.8.10 (Linux), Board:"WEMOS D1 MINI ESP32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

ESPAsyncWebServer.h: No such file or directory <-- any idea what I'm missing, this one is on line 50 of the .ino, I've searched for async in the libs and installed everything but still missing something.

DirkTas67, 2019-11-11:
@lpacchin 

ok 'kinda solved' it.

-> #include <NeoPixelBrightnessBus.h>
https://github.com/Makuna/NeoPixelBus

-> #include <ESPAsyncWebServer.h>
https://github.com/me-no-dev/ESPAsyncWebServer

completely nog idea if these are the right libs.

lpacchin, 2019-11-11:
@DirkTas67 
Yes I have **both** installed too

DirkTas67, 2019-11-12:
@lpacchin 

In the end I didn't get it to work this way, however on platformio I could make it work by filling in all the network information in the sketch and then flash the Wemos D1 mini ESP32, today I also found out that I could use Esphomeflasher in Ubuntu, and this works too by just flashing the binary's found here, so there must be something in esphomeflasher that makes it flash the right way

lpacchin, 2019-11-12:
@DirkTas67 
FUNtastic
Installed Esphomeflasher for mac and works smoothly (https://github.com/esphome/esphome-flasher/releases/download/v1.2.0/ESPHome-Flasher-1.2.0-macOS.zip)
It flashes all the partitions needed: bootloader, partition and bin, that I never able to do with esptool...to the esp32
Thank you!

closed 2020-01-07 by Aircoookie