wled/WLED #322: [Feature Request] Output to control a transistor on or off on a different pin based on E1.31 Control

jmoery, 2019-11-04:
Loving WLED and have had a great experience so far.  I am currently setting up holiday lights on my house and am wanting to do larger panels.  Currently I have a panel that is 5 rows of 21 LEDs each.  I want to hardware them so they are all running in parallel to use built-in WLED effects.  This looks pretty cool.  I also want the option to still use E1.31 on them and have them wired in series.  I was thinking I could just add extra data lines that connect the strips in series and turn on a transistor to have that line active when E1.31 is enabled, and off when it isn't enabled so you are basically able to control them all through software to be set in parallel or series.  

To make this work correctly, It seems I would need one additional data output pin that turns on when E1.31 is enabled and controls the necessary transistors to enable the series control and disable parallel control.  Is this something that could be easily implemented?

Aircoookie, 2019-11-06:
This could be done, but it seems like a too specific requirement to be implemented for all users.

Another solution (that doesn't involve hardware modifications!) would be to use 5 segments with 21 leds each, all running on the same FX. Currently only possible with the JSON API, but will be available in the web UI soon!

jmoery, 2019-11-07:
That's interesting.  Not having to do hardware changes would definitely be preferable.  I didn't realize there was a way to run different segments.  I'm assuming that the strands are all tied together in series and you are able to designate individual sections for local effect control?  That would be way easier.  Is there anything in the wiki on how to set it up?

brentbrooks70, 2019-11-07:
@jmoery 
>  Is there anything in the wiki on how to set it up?

Take a look at [API-Wiki](https://github.com/Aircoookie/WLED/wiki/JSON-API), but note that there have been some changes that may not be completely documented yet (as segment development is not complete).

jmoery, 2019-11-23:
I'm currently testing out segments with 0.8.5 and can't seem to get it to work.  I'm using this json message to try to create 2 segments but basically only the 2nd segment is getting applied for leds 0 through 22.  Just wanted to see if I'm doing something wrong.

"{ 'on': true, 'bri': 200, 'seg':[{'id':0, 'start':0, 'len':22, 'fx':2 },{'id':1, 'start':23, 'len':22, 'fx':3}] }"

jmoery, 2019-11-23:
> I'm currently testing out segments with 0.8.5 and can't seem to get it to work. I'm using this json message to try to create 2 segments but basically only the 2nd segment is getting applied for leds [...]
Nevermind, found the issue.  The color wasn't set in segment 2 so it wasn't turning on obviously.

closed 2020-04-05 by stale[bot]