wled/WLED #323: [-Feature Request-] Lighting sequence preview window/effect listed

SirRedZ, 2019-11-05:
Just was thinking that I couldn't look out my window and see what effect is currently playing. A small preview window displaying a sample of the current pattern would be kinda cool?

Just one idea alongside many others!

[-=SrZ=-]

Aircoookie, 2019-11-06:
Thanks to @photocromax 's help, this will be available soon, maybe even in 0.9.0!

LADGRES, 2019-11-07:
I was just going to add this enhancement as well.  Even just some simple text to show what settings are currently being used (ie.  Primary Color, Secondary Color, Preset No., FX Mode, FastLED Palette)

Aircoookie, 2019-11-09:
For those interested, here is the solution by @photocromax !
https://github.com/photocromax/WLED-live-visualizations/
Will become part of WLED shortly :)

Tollbringer, 2019-11-09:
This looks very promising.... [@Aircoookie ] don't forget find & replace doesn't change browser title version number!!!! 😅  

Great work and thank you.

photocromax, 2019-11-09:
For the ones doesn't want to mix the projects files, I've prepared a bin file compiled for D1 mini/nodeMCU (4MB no SPIFFS) default led pin (D4)
https://github.com/photocromax/WLED-live-visualizations/tree/master/bin 

Be aware you can encounter some minor issues.

photocromax, 2019-11-09:
> This looks very promising.... [@Aircoookie ] don't forget find & replace doesn't change browser title version number!!!! 
> 
> Great work and thank you.

I forgot to update it too. :P another minor issue

Tollbringer, 2019-11-09:
I tried !!!... lol.

Question: I am currently using WLED on a 1MB ESP controller (we've discussed in the past the need to use ESP-1M binary. It is a Magichome Controller. Will I be able to take advantage of the preview feature on this device? I have purchased a bunch of them for my house to all run WLED.

Thank you.

Will

photocromax, 2019-11-10:
> I tried !!!... lol.
>
> Question: I am currently using WLED on a 1MB ESP controller (we've discussed in the past the need to use ESP-1M binary. It is a Magichome Controller. Will I be able to take [...]

Tomorrow I'll try to compile a version for 1MB module, which GPIO do you use for the led strip?

Tollbringer, 2019-11-10:
The standard one. (GPIO2 - correct?)

Will

photocromax, 2019-11-10:
> The standard one. (GPIO2 - correct?)
> 
> Can you please enable Home Assistant auto-discovery also? TY!!!!
> 
> Will
Sorry, I forgot you don't have so much choice with and ESP-01.
You can try with the new bin in the repository (generic.1M)
Remember: NO OTA update!!! 
The new features are only in the mobile UI.
HA discovery is enabled.

Tollbringer, 2019-11-10:
Thank you. So I must serial flash this one? No AT update pertains to this update, not future updates?

photocromax, 2019-11-10:
I don't think this feature will fit in 1M with OTA enabled. Let's wait for when it will be fully integrated and optimized. 
If you wanna try it advance, please, be ready to serial flash the module, or just flash it on a more capable module. Those are purely UI features, it doesn't worth to dismantle something if it's not easy and comfortable to serial flash it.

photocromax, 2019-11-10:
> Thank you. So I must serial flash this one? No AT update pertains to this update, not future updates?

I've uploaded a new version with OTA enabled, Mobile UI (with the new live visualization).  HA discovery is still enabled.
Visualization is only on mobile UI (right now). 
I don't have any ESP01 to test if the OTA process goes right. If it's not too hard to serial flash your device you can use this bin.

Tollbringer, 2019-11-10:
My module is not the ESP01 - the magichome controllers use 8285 processor and Im struggled serial flashing wled, i get it to flash, and the original page comes up, but after wifi entry it dissapears forever

Aircoookie, 2019-11-10:
@Tollbringer 8285 is just ESP8266 with 1m embedded flash, that means it should behave exactly like an ESP01. If it "disappears", it should actually be connected to your network. Does it shouw up in your router device list or when running a discovery in the WLED app?

photocromax, 2019-11-11:
> My module is not the ESP01 - the magic home controllers use 8285 processor and I'm struggling serial flashing wled, i get it to flash, and the original page comes up, but after wifi entry it disappears forever

You can find another bin for esp8285 with OTA enabled and Mobile UI disabled (to save up some space). Live PreIview is now available in classic UI

Tollbringer, 2019-11-11:
Hello my friend, where do I find this binary? I just checked the releases and don't see anything (released 0.86 17 days ago)

photocromax, 2019-11-11:
> Hello my friend, where do I find this binary? I just checked the releases and don't see anything (released 0.86 17 days ago)

Here (same link as before)
https://github.com/photocromax/WLED-live-visualizations/tree/master/bin

This feature is not already part of WLED, that why are in a different repository

Tollbringer, 2019-11-12:
@photocromax @Aircoookie 

Great News ! I'm back ! I tried again with my Magichome controller today, and I was able to flash it with photocromax 0.86-es8285 binary. 1 flash, configured wifi settings, rebooted. then configured static IP, rebooted, and back in business. Only think I noticed is mobile UI not enabled on this binary. I am not sure how to approach home assistant auto-discovery yet.. But thank you very much Gentleman. Really. Thank you.

Will

photocromax, 2019-11-12:
> @photocromax @Aircoookie
> 
> Great News ! I'm back ! I tried again with my Magichome controller today, and I was able to flash it with photocromax 0.86-esp8285 binary. Worked on 1st flash, configured wifi settings, rebooted. then configured static IP, rebooted, and back in business. Only think I noticed is mobile UI not enabled on this binary. I am not sure how to approach home assistant auto-discovery yet.. But thank you very much Gentleman. Really. Thank you.
> 
> Will

HA discovery is enabled. I don't use HA, but I think you just have to configure the same MQTT broker for WLED and HA

You can visualize the effect preview in the classic UI. I hope it works fine

closed 2020-01-07 by Aircoookie