wled/WLED #324: Is there a way to cycle all the FXes and Pallettes and intensities and speeds in a random way?

l0c0luke, 2019-11-05:
I am spending hours playing with the combos and I find they are all amazing. So instead of creating presets and cycling through them all to get a random experience, is there a way to just let the luck of the draw happen?

sansillusion, 2019-11-05:
That would be interesting !

recliq, 2020-01-09:
What about a small UserProgram...
`wled06_usermod.ino`:
```/*
 * This file allows you to add own functionality to WLED more easily
 * See: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality
 * EEPROM bytes 2750+ are reserved for your custom use case. (if you extend #define EEPSIZE in wled01_eeprom.h)
 * bytes 2400+ are currently ununsed, but might be used for future wled features
 */

//Use userVar0 and userVar1 (API calls &U0=,&U1=, uint16_t)

long lastTime;
int delayMs;

//gets called once at boot. Do all initialization that doesn't depend on network here
void userSetup()
{
    userVar0 = 0;
    userVar1 = 0;
    lastTime = 0;
    delayMs = 0;
}

//gets called every time WiFi is (re-)connected. Initialize own network interfaces here
void userConnected()
{

}

//loop. You can use "if (WLED_CONNECTED)" to check for successful connection
void userLoop()
{
    delayMs = userVar0 * 1000;

    if (delayMs > 0 && millis() - lastTime > delayMs)
    {
        lastTime = millis();
        userVar1 = 1;
    }

    if (userVar1 > 0)
    {
        effectCurrent = random8(0, strip.getModeCount() - 1);
        effectSpeed = random8(0, 255);
        effectIntensity = random8(0, 255);
        effectPalette = random8(0, strip.getPaletteCount() - 1);
        colorUpdated(1);
        userVar1 = 0;
    }
}
```
Set `&U0=<seconds>` to randomize the effects periodically each x seconds or 0 to disable,
set `&U1= 1` to trigger randomize effect once.
You may also maybe limit the value range for speed and intensity...

closed 2020-05-15 by stale[bot]