wled/WLED #325: Suggestion, not an "Issue"

bphick, 2019-11-05:
Would it be possible to put an indicator in the description of those Effects that utilize multiple colors?
Like a * perhaps?   ...And at this point, alphabetization of the Effects, etc?

TheZoker, 2019-11-05:
Partly a duplicate of:
https://github.com/Aircoookie/WLED/issues/292

And a general tip for github: Use meaningful issue titles like "Highlight effects with multiple colors" or "Sort effects alphabetically", something like that 😉

Aircoookie, 2019-11-06:
Thank you for the suggestion.
Likely won't be adding this one for some time since there are higher priority additions/issues to be worked on. It's a good idea, but you can easily figure out which effects use palettes (almost all of them do if you select something other than "Default" by the way)
But I'll keep it in mind :)

brentbrooks70, 2019-11-08:
@Aircoookie Is it possible to add an "active" indicator to the mobile UI? I guess like you did for the power and sleep buttons. Personally, I prefer dropdown lists myself but I know lots of people do not.

[image]

Aircoookie, 2019-11-09:
@brentbrooks70 You mean highlighting the effect/palette currently selected?
I kind of wonder why i have not considered that for the current UI honestly. New UI will have it!
About the selection itself, I am really debating between the "div with one button for each effect/palette" (like in the current mobile ui) and the dropdown approach. The advantage of dropdown is that you immediately see the set effect, but the drawbacks are that you need one more click each time you want to change the FX and that the dropdown UI is inconsistent/depending on client OS.

brentbrooks70, 2019-11-09:
@Aircoookie That is exactly what I meant, seeing which FX or FP is active by being highlighted. Glad you think it would be useful! I have also seen great discrepancies in dropdown lists between OSs and even different browsers on the same OS which is why I had said lots of people do not like using them.  The work by [JasonCoon](https://github.com/jasoncoon/esp8266-fastled-webserver) does a good job handling this using bootstrap.mini.css. The dropdowns look and behave the same from
Android browser 
Chrome on Android, IOS, Windows
Safari on IOS
IE and Edge on Windows, cannot speak for MacOS as I do not have one of those anymore.

blazoncek, 2022-02-21:
UI FX and FP highlighting has been implemented in 0.12 and refined in 0.13.
Additionally there is experimental implementation of UI control in my fork (which was ported from SondReactive fork). Some fundamental code has already been implemented in WLED master as well. PR for UI control for master is in the review process.
Closing.

closed 2022-02-21 by blazoncek