wled/WLED #326: Grouping LEDs

Sprobin, 2019-11-05:
Hi,
Is it possible to group Leds? 
Want to build something like the nanoleaf Aurora lamps. For that I need to group 3 leds, so that they always have the same color. Is it possible to that? I couldn't find a setting for this.

brentbrooks70, 2019-11-05:
This will be available in ver 0.90 via “segments”. You could also use ws2811 LEDs, they are 12v instead of 5v but they have 1 controller per 3 LEDs so they are always grouped in 3s. In other words, if you set pixel1 to red and pixel2 to blue you would actually get 3 red then 3 blue LEDs, you would of course need to set the number of LEDs to 1/3 of the actual number

Sprobin, 2019-11-05:
Thank you very much for the fast and helpful answer.

closed 2019-11-05 by Sprobin