wled/WLED #327: E1.31 memory issue

CaffeinatedFunctionality, 2019-11-05:
I prefer WLED over ESPixelStick, unfortunately for my XLights this year I have to revert back to ESPixelStick at least until I can resolve this issue. 

When using WLED with E1.31. It seems to work on very small sets. When having one WLED nodemcu running with 460 LEDs it seemed to work perfectly on unicast. Multicast seemed to only get the first universe, but couldn't seem to see any more than that. When adding another WLED nodemcu(running unicast) of another 494 pixels it seemed to be sluggish and both seemed to randomly reset the chip which showed the solid initial color it gets after being reset. I'm assuming it might not be able to handle up to 600 leds like ESPixelStick. Maybe this is a memory issue? The app also showed it randomly disconnecting from the wifi which may be it resetting or just being overloaded. I'm not really sure. It was odd that it worked on one chip but not when using more than one. As far as I know unicast makes it not have to process the universes it doesn't care about so I would have assumed this would be more memory efficient. Maybe a bad assumption, but I think there has to be a memory issue here with this many leds and E131

yusufsulaiman, 2019-11-06:
I have tried it, for 300 LEDs it's ok, but for 400/500/600 LEDs there are delays
https://www.youtube.com/watch?v=MK9M_R8qT3c

CaffeinatedFunctionality, 2019-11-06:
That looks amazing. I hope I can figure it out. 

⁣Sent from BlueMail ​

hristo-atanasov, 2019-11-13:
@yusufsulaiman I have the same problem. I'm glad that it isn't only me .. It happens with 6-7 wled-ed NodeMCUs with 180 pixels each .. I'll have to revert to ESPixelStick temporary until this is fixed. I love WLED, but I don't know how to resolve this issue by myself.

hristo-atanasov, 2019-11-13:
Hm .. Now I have another thought ..What if set the processor speed to 160mhz .. And lwip to 2 Higher bandwidth .. I'm not sure if this can help .. Also when uploading we have to use DOUT ..

thedannymullen, 2019-12-19:
I have reverted back to pixel stick for same reasons listed here.  
After the season I am going to try to play with wled I am hoping it can be fixed as I like wled also!!

Aircoookie, 2019-12-19:
Sorry for not getting back to you on this issue, guys! It somehow flew under my radar, but E1.31 performance should be significantly improved in the latest 0.9.0-b1 version since I am using the async library now.

Happy holidays to you all :)

thedannymullen, 2019-12-21:
Thanks I will check it out after holidays!  I want to do so debug when I have a min.

closed 2020-01-05 by Aircoookie