wled/WLED #328: WiFi Not Connecting

KB706, 2019-11-06:
I am generally new to coding of any kind so please be patient with me. I have a NODEMCU Chip I have Flashed with ESP8266 to use WLED to control my LEDS. When I get to the WiFi settings portion, I add my router name & password, then hit "save and reboot"... It then gets stuck in the screen saying "WiFi Settings Save - Rebooting Now..." Never goes anywhere from there. When I try to find the device on the Phone app, it is able to find it but says it is offline. 

What am I doing wrong? Help...

I've tried unplugging it and replugging it back in after flashing it. 

Nothing I do seems to change anything. Please guide me in the right direction. I am trying to put together a nice holiday display for my 2yr old. 

Thank You in advance for all the help.

raoulteeuwen, 2019-11-06:
Can you please check the settings you used to flash the NodeMCU in detail, https://github.com/Aircoookie/WLED/wiki ? Anything different from the recommended settings? And what binary did you use?

DuschdrBabbe, 2019-11-06:
Hi,
had the same issue... try the wled_0.8.6_ESP8266_lwip1.bin this will work.
I think there is a problem with the WLED_0.8.6_ESP8266.bin

Greetings...

KB706, 2019-11-06:
> Can you please check the settings you used to flash the NodeMCU in detail, https://github.com/Aircoookie/WLED/wiki ? Anything different from the recommended settings? And what binary did you use?

I used esp- flasher program and flashed the file "wled_0.8.5_esp8266.bin. I didn't modify anything. As for what binary... I'm not sure. I'm very new to this side of things.

raoulteeuwen, 2019-11-06:
> I used esp- homeflasher program and flashed the file "wled_0.8.5_esp8266.bin. I didn't modify anything. As for what binary... I'm not sure. I'm very new to this side of things.

I'm not familiair with that flasher. But before looking at other things, please try flashing the file @DuschdrBabbe mentions and let us know whether that fixes your problem.

KB706, 2019-11-06:
> I'm not familiair with that flasher. But before looking at other things, please try flashing the file @DuschdrBabbe mentions and let us know whether that fixes your problem.

I just tried with no change. It's odd. The smart phone app can't "find" it either. But when I close the cancel the discovery and go back to the main page, it's there. But it says offline.

I even tried a different chip thinking the NODE may have been faulty. Still the exact same results.

DuschdrBabbe, 2019-11-06:
After Setting your WIFI Settings in WLED you need to connect your phone back to your Home wifi... also just reboot the mcu again... wait a couple of seconds the try again... have you also checked the ip of the Node? Try connect to the nodemcu via a normal browser...  can you see  the nodemcu in your router?

KB706, 2019-11-06:
> Can you please confirm that you've redone the flashing, using file WLED_0.8.6_ESP8266.bin ?

I started with the "WLED_0.8.6_ESP8266.bin". That one didn't work. So I tried the 8.5 too. No such luck.

KB706, 2019-11-06:
> After Setting your WIFI Settings in WLED you need to connect your phone back to your Home wifi... also just reboot the mcu again... wait a couple of seconds the try again... have you also checked [...]

I have been doing this on my laptop. Just tried to discover the chip on my phone. The laptop automatically goes back to the Home Wifi... But when I try to the "http://wled-"NAME".Local" it times out saying it took too long to respond. 

When I login to my router, I don't see the NodeMCU in the list of connected devices...

raoulteeuwen, 2019-11-06:
> > Can you please confirm that you've redone the flashing, using file WLED_0.8.6_ESP8266.bin ?
> 
> I started with the "WLED_0.8.6_ESP8266.bin". That one didn't work. So I tried the 8.5 too. No such luck.

If you go to https://github.com/Aircoookie/WLED/releases and look at the assets, you see one named "wled_0.8.6_ESP8266_lwip1.bin".  That's the one @DuschdrBabbe and i would love you to try and inform us whether that makes any difference.

KB706, 2019-11-06:
> Can you please check the settings you used to flash the NodeMCU in detail, https://github.com/Aircoookie/WLED/wiki ? Anything different from the recommended settings? And what binary did you use?

I used esp- flasher program and flashed the file 

> If you go to https://github.com/Aircoookie/WLED/releases and look at the assets, you see one named "wled_0.8.6_ESP8266_lwip1.bin". That's the one @DuschdrBabbe and i would love you to try and inform [...]

That is the one I tried earlier after reading the comments this morning. No change.

chris89178, 2019-11-07:
I tried version WLED_0.8.6_ESP32.bin, it´s still not saving the changes.

ChicJr, 2019-11-07:
Hello all,

I'm experiencing the same wifi connection issues as KB706.

I just flashed "wled_0.8.6_ESP8266_lwip1.bin" onto the NodeMCU and now the NodeMCU is connecting to my WIFI network, but the autodiscovery on the WLED Android app doesn't discover it. 

I then added it manually with the IP address, and the app appears to think it is valid, but it permanently sees it as Offline.

I've tried this on 2 NodeMCUs (both from the same manufacturer).

KB706, 2019-11-08:
> Hello all,
> 
> I'm experiencing the same wifi connection issues as KB706.
> 
> I just flashed "wled_0.8.6_ESP8266_lwip1.bin" onto the NodeMCU and now the NodeMCU is connecting to my WIFI network, but the autodiscovery on the WLED Android app doesn't discover it.
> 
> I then added it manually with the IP address, and the app appears to think it is valid, but it permanently sees it as Offline.
> 
> I've tried this on 2 NodeMCUs (both from the same manufacturer).
> 
> **UPDATE**: After (3) Powerdown, restart and restart Android app sequences, the Android app seems to be OK connecting to the WLED NodeMCU now. I've successfully powered down and rebooted the NodeMCU twice since, and the app finds and controls it each time.
> 
> Thanks!

So that's good to know. Maybe I'll reflash it with that and so multiple restarts and see what happens. Thanks for the input.

KB706, 2019-11-08:
So I did as ChicJr mentioned. With mild success. I say mild only because the connection seems to be spotty for some reason. All other devices connected show to have a strong connection. 

Sometimes, on the App I can bring up the Nodemcu. But others it just gets stuck saying loading Wled IU. It also goes in and out showing "offline", but it will let me adjust the brightness...

brentbrooks70, 2019-11-08:
@KB706 This really sounds like bad WiFi signal/connectivity. If you are on a PC can you open a CMD prompt (powershell would work as well) and type:
**ping** _DeviceIP_ **-t**
if you get continuous replies with decent time (ms <100) then it's probably not signal strength

KB706, 2019-11-08:
> @KB706 This really sounds like bad WiFi signal/connectivity. If you are on a PC can you open a CMD prompt (powershell would work as well) and type:
> **ping** _DeviceIP_ **-t**
> if you get continuous replies with decent time (ms <100) then it's probably not signal strength

I will try this when I get home from work. I tried setting the lights up directly next to the router and it did the same thing.

So, if I do find that the signal strength is the problem what do I do from there. All my other devices have no issues.

fribse, 2019-11-17:
And you di have 2.4 GHz wifi, right? The esp can not connect to 5GHz.

KB706, 2019-11-17:
So, after refreshing "wled_0.8.6_ESP8266_lwip1", Unplugging it the NodeMCU after each change(once after flashing it & once after connecting it to wifi) and plugging it back in, it eventually started working. Though once connected to the lights it took a couple power cycles to get them to connect.

Side Note: I also assigned it a static IP address as well.

atxbyea, 2019-11-22:
I started using WLED today and I too don't get it to save WiFi settings.

On 0.8.5 it would never connect to the WiFi, on 0.8.6 it connects but after a restart it is back in AP mode.

Running on wemos d1 mini, flashed with esp8266flasher.exe

Everything works fine in AP mode

Trebor-Eraeps, 2019-11-29:
wow this genuinely sucks, had everything working back in october, thought i was cool this weekend to install my new year around holiday lights and wow, cant connect just like everyone else here.  CRAP!!! has anyone found a solution, tried all of the above, not of it works.  i must've spend 8hrs trying to reconnect several brand new esp8266's thinking it was a hardware issue.

Mariu86, 2019-11-30:
You must explain what you did to understand what is happening.

fribse, 2020-03-29:
@Mariu86 and you are sure it's trying to connect to your 2.4 GHz wifi? I have currently seven running here on WLED 0.9.1, and so far they work very very well.

kombi21251, 2020-04-04:
Hello! I have the same problem.
Changing the settings of the router helped me.
I am changed the channel of wireless network from "auto" to "1".
My router is TP-LINK AC750.
I am use WLED_0.8.6_ESP8266.bin

But not for long. After some time, WLED disconnected: / I select another channel in the WLED settings, and set the automatic channel on the router. WLED connect. But I do not see the WLED IP device on the router. Despite this, WLED is available on 192.168.0.100

closed 2020-08-09 by stale[bot]

DontDoThat21, 2021-01-17:
Still an issue after trying all above steps above.

rjworks, 2022-07-27:
help