wled/WLED #329: open page after connect to home networ-no open

pepeEL, 2019-11-06:
Hi
I write to my nodeMCU this file: WLED_0.8.6_ESP8266_ledpin4.bin
Then i search network WIFI AP and found WLED-AP. Connect to it by default pass. Then i setting to connect to my home networ... setting SSID, pass and leave 0.0.0.0 for IP to get DHCP.
Save and connect and nodemcu connect to my home network. On my router i see that he get IP like 192.168.xxx.xxx.
I try open this address in web browser but no open. How i can go to this device to configure it ?

brentbrooks70, 2019-11-08:
Can you ping the reported IP Address? Are you trying to connect from a wireless device or wired? maybe look at [ChicJr's UPDATE comment](https://github.com/Aircoookie/WLED/issues/328#issuecomment-551292536)

pepeEL, 2019-11-08:
I am trying connnect from device wire... not test ping. Then write firmware WLED_0.8.6_ESP8266_ledpin3.bin and on this page is open. But when again back to WLED_0.8.6_ESP8266_ledpin4.bin again no open page.

pepeEL, 2019-11-08:
On firmware WLED_0.8.6_ESP8266_ledpin4.bin i can not ping device.

pepeEL, 2019-11-08:
I write on other NodeMCU but now v3 and firmware works ok. Only on NodeMCU v1 not working correct

closed 2020-04-05 by stale[bot]