wled/WLED #330: ERR_ADDRESS_UNREACHABLE combile/app

T3rry12, 2019-11-08:
So i set up half a dozen of these controllers on wemos D1 Mini, at first they all worked fine in the app and on mobile but now every time i try to use the app it says the device is offline and gives the above error webpage. It works fine on my desktop PC but not on app or on android browser. I am on the same network with all devices,
[image]
[image]

brentbrooks70, 2019-11-08:
@T3rry12 can you ping that IP from another wireless device/computer? if not, I would look at your wireless router/access point to see if anything like:
device isolation, network isolation, client isolation, AP isolation or anything similar 
has been enabled by a recent update, this would prevent any wireless device from talking to another. 
I use an android app called Network Toolbox (by APPSROPOS) to do simple pings and network scans from phones if you do not have a wireless PC.
Good luck!!

@Aircoookie I think I read somewhere you were looking at adding RSSI to JSON/Info, would it be possible to add IP Address as well to assist with network connectivity troubleshooting? Maybe it could spit it out serial as well upon initial connect so non-bonjour machines know the IP immediately after flashing/rebooting?!?

T3rry12, 2019-11-08:
thanks for the help, a lot of that went over my head, i have an edgemax lite so anything specific to that would be appreciated.

brentbrooks70, 2019-11-08:
@T3rry12 I just looked through what I think is the manual for your device,
https://dl.ubnt.com/guides/edgemax/EdgeRouter_Lite_UG.pdf
but I did not see anything about isolation. Do you have a guest network SSID defined as well as your main SSID? It looks like when guest networks are enabled it adds client isolation to all WiFi networks. You would need to create some sort of loopback adapter on the device or disable Guest access. 
I would love to help more but I am not sure walking somebody through creating such an interface via message board would be the best idea. If we break communication you would not be able to reply for additional help, SORRY!

T3rry12, 2019-11-08:
I appreciate your suggestions thanks, I guess i have some research to do :)

Thank You

closed 2019-11-08 by T3rry12