wled/WLED #331: Macro Mayhem!

brentbrooks70, 2019-11-08:
@Aircoookie (or anybody) 
I have another strange issue, this time with Macros...
If I define a macro such as:
_T=1&A=4&FX=65&FP=30&SX=0_
nothing after the A gets set. In other words, the above macro would turn on the light, set the brightness to 4 and that is it.
If I move it towards the end of the macro as in
_T=1&FX=65&FP=30&SX=0&A=4_
everything gets set except the A (brightness).
Now, if I set MACRO 1 to
T=1&FX=65&FP=30&SX=0&M=2
and set MACRO 2 to
A=4
it works! Am I missing something? Doing something wrong? I would hate to have to use 2 macro positions for each setting. :( Sorry to bombard you with so many questions tonight. 

This is a binary (WLED_0.8.6_ESP8266.bin) flashed LoLin NodeMCU V.3 ESP8266MOD AIThinker
Same board I was using for [issue 190](https://github.com/Aircoookie/WLED/issues/190#issuecomment-551357994) earlier this evening. Thanks again!!!

All Hail @Aircoookie!!! Long Live WLED!!!  
Okay, its obviously late and I am getting stupid, time for bed.

Aircoookie, 2019-11-09:
Hi!
Thank you! I'm very happy you like WLED so much!
If you are using A=, it will always turn the LEDs on (or off in case of A=0), so using T=1 in that case is not required. In fact, the T will override the A, causing the brightness to be set to whatever it was before the LEDs were turned off.
I don't have a clue why the other commands haven't worked in the first macro you tried but i believe it may be related to the segment configuration you have reported in #190.
:)

brentbrooks70, 2019-11-09:
Excellent, @Aircoookie you are 100% correct, combining T & A causes conflict
FX=65&FP=30&SX=0&A=4
or
A=4&FX=65&FP=30&SX=0
both produce expected results.
Thanks once more!!

closed 2019-11-09 by brentbrooks70