wled/WLED #332: Manual how connect it with strip led and NodeMCU

pepeEL, 2019-11-08:
Hi
I am beginner. And start with home automation. WHere i can find some manual how connect led stips with NodeMCU with writed firmware WLED ? Please help me....

raoulteeuwen, 2019-11-08:
Not sure what you're looking for: info to get started with LEDs, info to get started with home automation, or info on how to integrate LEDs with home automation. But i've documented the best sources i found that helped me get started with LEDs at https://raoulteeuwen.blogspot.com/2019/08/starting-with-led-strips.html .

I'm using Home Assistant on a Rpi3 for home automation. Don't have EN write up on getting started on that, but JuanMTech has many nice video's on that, like https://www.youtube.com/watch?v=qnCRcGTznXs .

pepeEL, 2019-11-08:
But When i have nodemcu with WLED i must only connect wire D4 from nodemcu to Led strip and ground and power from power supply to Led strip?

raoulteeuwen, 2019-11-08:
Did you check the links i supplied? Anyway, you also need ground from your NodeMCU to the LED strip, and that ground needs to be shared with the ground between your power supply and the LED strip. Like the schematic on https://www.planet4.se/led-strips-in-home-assistant/

sjude68, 2019-11-14:
> But When i have nodemcu with WLED i must only connect wire D4 from nodemcu to Led strip and ground and power from power supply to Led strip?

You will not need another power supply for led strip as you can use the 5v and gnd pins of the nodemcu for the strip. Power the nodemcu by usb or same 5v and gnd pins.

roguestreak, 2019-11-21:
Don't forget to pay attention to the Data line on the strip, one end is "in" ("DI") and one is "out" ("DO"); I stupidly forgot all about this and spent quite some time trying to figure out what was wrong, only to find I'd connected to the "out" end.
I just got WLED running on a nodemcu last night, works fine for a 150LED strip directly from the board (am using a 2A USB supply)

pepeEL, 2019-11-25:
I connected my POWER SUply to my nodeMCU and my LED strip. NodeMCU works ok but no control my led strip. Light only first 8 first led on strip. Other led from lenght 5m not working. But when try control by WLED no reaction.... Where is a problem ?
Power supply is 5V 70A

roguestreak, 2019-11-27:
So can you access the WLED interface?  If so, under settings how many LEDs have you got it set to?
What colour are the 8 LEDs showing up as, and are these 8 responding to commands?

pepeEL, 2019-11-27:
Problem was solved. I flashed my nodemcu by other flasher and then set how many pcs LED and it start working correct.

closed 2020-01-07 by Aircoookie