wled/WLED #338: Is it possible to link multiple nodes of WLED together?

jerryngm, 2019-11-10:
Hi,
I know that WLED can sync multiple WLED nodes, however is it possible to combine them? 

E.g.  I have 2 separated WLED server each has 30 LEDs (Running on ESP32). Can I combine them to create a server with 60 LEDs all together? 

Thanks :)

Aircoookie, 2019-11-12:
Hi,
right now this is not possible, unfortunately (i'm assuming you would want this feature to have a smooth, seamless motion, for example have the wipe effect behave like if the 60 LEDs were connected to just one ESP).

However, I will try and find a way to achieve this :)
For example, it could be possible to set a virtual LED count (using segment bounds) higher than the actual LED count together with an offset to render the effect on LEDs 1-30 (of the 2nd unit) as if they were LEDs 31-60!

Aircoookie, 2019-11-12:
This might also actually be useful for people who want a loooong strip (like 3000 LEDs), which is not possible with just one ESP due to hardware limitations. This feature could then make for example 3 units of 1000 LEDs each behave like one large one!

buildersacademy, 2019-11-13:
Thanks for considering my idea :)

Do you think that we can utilise something like ESP-MESH for this?

jerryngm, 2019-11-13:
Thanks for considering my idea :)

Do you think that we can utilise something like ESP-MESH for this?

brentbrooks70, 2019-11-14:
@jerryngm esp-mesh (I presume you are referring to esp-now) requires dedicated use of the WiFi radio and would prevent direct control via GUI, HTTP API or MQTT API of the master device. If you needed/wanted to use this in an area with no or limited wifi connectivity you could just have the master device be an AP and all others connect to it, might even auto register via wifi to configure itself... this way you could still connect to the AP and control all devices behind it. You cannot connect to an espnow network with anything other than an esp device, at bnb least as far as I know

tonyn0, 2019-12-31:
Watching!

DieNummerEins, 2020-01-21:
Hi Aircookie, did you make already some development on this feature or at least some basic considerations?

Especially I'd like to know, if combining would make it still possible to reverse the led-order ("reverse led order (rotate 180)" in led preferences)?

I'm currently installing led-stripes (sk6812, rgbww) in every room of my house as indirect light. It's a diy construction with ikea vidga (curtain rail, so i can hang curtains or decoration), cabel canal and other stuff from hardware store. At the end the led-stripe is stick to alu rail (45 degree to ceiling) I have to decide where to put the controller (in my case esp32). In other words: Where should every stripe start and end? In one room I have 7x stripes, each stripe is 4-5m long. Once the led-stripe is installed it's hardly possible to turn the direction because the self-adhensive function doesn't like that.

If reverse mode will be possible, then it doesn't matter where to place every controller and to start the correspondent stripe so effects like wipe would run correct. But if not, I should start every stripe where the previous ends.

btw: Awesome project and work! THANKS!!!

Aircoookie, 2020-01-21:
@DieNummerEins hi! I'm very happy that you like WLED so much :)

That sounds like a very cool project! I do not yet 100% know how I am going to implement this feature, but I can promise you that the "Reverse entire strip (rotate 180)" setting will work for each individual strip, thus you have more freedom for your controller placement :)

ncgbrown, 2020-02-24:
Thanks for all the hard work!  I am connecting two sets of undercounter leds in my kitchen that are physically separate.  Therefore I require two controllers.  Having them behave as one long strip would be preferable.

+1 for this feature - and again, thanks

SWATr, 2020-04-14:
Love WLED, I would also like to see this feature added.

KensonN, 2020-05-22:
Would also love this!

strfngr, 2020-05-24:
This! I'd be highly interested in such a feature.
Just about to install ~1700 leds with 4 ESP8266s around my living room ceiling and it's a bummer many of the great effects won't work well (chasing, wipe, and many more).
Any news on the timeline?
Alternatively: Would be a very bad idea to just run the 1700 leds from one board?

Legsmaniac, 2020-05-24:
> Alternatively: Would be a very bad idea to just run the 1700 leds from one board?

I'd say no, not a bad idea at all. I'm running 2040 leds on one ESP8266 with no problems whatsoever.

StimpyMGS, 2020-06-08:
I've subscribed to this thread because I'm also interested to see this work. However, my use case would be slightly different. I'm using Hyperion which is a piece of software that grabs the edges of my desktops video output and translates that to the leds that I have attached to the back of my monitor running WLED, aka ambilight. Now I would like to have that same output on another WLED driven strip that I have elsewhere in my room, to even submerge more in my game.

justinoreilly, 2020-06-17:
Another +1 for this feature. I was specifically looking to do large chase and wipe effects along my fence ~100ft per side. With 60 LED/Meter this turns out to be 30.5 meters or 1830 LEDs per side with a total of ~3500 LEDs. Sadly more than WLED currently handles.

siemaszka, 2020-06-20:
hi. can you make esp8266 more than one line to ws2815?
it means:
1 line=180pix
2 line=180pix
3 line=180pix

Aircoookie, 2020-06-20:
@siemaszka if you want them to all display the same thing, you can wire the inputs in parallel to the same esp8266 pin. If you want them to display different effects, currently putting them in series is the only option, but that will change soon. Even so, 3 outputs on ESP8266 might be hard to achieve, we'll see!

PeteRepeat77, 2020-08-01:
I'm assuming this hasn't made any progress yet? Currently I'm using tasmota firmware on two wemos mini and controlling two led strips only with hubitat as the interface/UI. For ambient lighting this works fine however I am curious about better affects if I move to outside for Xmas lighting.

So early days of ws2812b strips for me.. However to move outside with affects it seems wled is a big step up from just using the tasmota firmware. It also seems there is a driver for hubitat as well so I can encorporate it into my home automation. However not being able to make an affect go across multiple strips is something I'd like to do when I make it outside. I realise syncing the same affect is now possible with led but that only starts them all at ye same time yes? Shame there isn't an offset that can be entered so if an effect is a consistent timing then the offset can be adjusted to make it almost aeemleas between strips.

Aircoookie, 2020-08-03:
Not yet. This is planned to be introduced together with multi LED pin support hopefully in time for the holiday season :)

rawframe, 2020-10-29:
Do you think this will be ready in time for Christmas?
If not will we have more than 10 segments available in the next release? Or if not possible can it be introduced for esp32 only.

I also have the need to span effects across multiple wled devices. At the moment I am power injecting a very long 'virtual' strip by bridging each spaced apart real strip with a data cable. This means more cables and also the data cannot go very far without needing to regenerate.

On the front end, perhaps we could select one wled as a master which has a 'virtual/hybrid segment' composed of the ids and segments made available from slaves wleds.

ejdprops, 2020-10-30:
> Do you think this will be ready in time for Christmas?

I’m looking at using multiple WLEDs for my holiday lighting as well (though, not this year...no time!).  Since WLED can be controlled by DMX lighting protocols..I’m planning to use xLights.  You can easily make multiple WLEDs act as one strip based on how the segments are defined.  Dr. ZZ has a great video on using WLED for holiday light shows

rawframe, 2020-10-31:
@ejdprops Thanks for the reply. What do you mean by 'based on how the segments are defined"?
@Aircoookie Ultimately it would be nicer to not need the use of another controller to achieve this.

Luvz2Fly, 2020-11-30:
Just started using WLED after I gave up a few months ago trying to do similar stuff using MQTT and running into issue after issue for months and months (when I started I had hair!). Got WLED up and running in MINUTES and does everything I need to do (except this one topic). so first I want to thank the developer and all the contributors for an awesome project, so far it's amazing!!

I just want to add another voice to what is being said here. As previously mentioned by others I have under lighting cabinet that spans across a large kitchen and was not pre-wired easy so I need at least two maybe three nodes that I hope to run as a single unit.  Hopefully this can be added feature, otherwise I'll just have to turn each one on individually I guess for now but still much better than where I was months ago!!!

Thanks again,. All of us here can definitely appreciate and recognize the hard work that went into making this program perform as well as it does!!

krizzo, 2021-08-05:
A method of processing would be to create a server/client setup and use NTP syncing along with a buffer/queue/stack on each client. So technically the server is rendering 1-2 seconds ahead and sending frames out to each WLED node. Each node just displays the "live" frame then discards it based on the time.

steppel, 2021-12-12:
Are there by any chance any tutorials for bundling multiple esp to have one continues virtual strip using the implemented ddp?

tonyn0, 2021-12-12:
All you do is choose one main ESP and add virtual LEDs like you're adding pins.

steppel, 2021-12-13:
> All you do is choose one main ESP and add virtual LEDs like you're adding pins.

Can you be a bit more precise please? My end goal is, having let's say, 3 nodes act as if it was one strip. An effect starting at C1 (cabinet 1) moving to C2 and then to C3. I don't want them all mirrored.

bkaufx, 2021-12-13:
You do it under LED Preferences / Hardware setup / LED outputs.  Here is a picture where I have an LED strip connected to GPIO and 5 bulbs connected by DDP:

[image]