wled/WLED #340: [Feature request] Light panels

MajorLOLGaming, 2019-11-11:
It would be cool to have support for custom light panels (like nanoleaf panels) where you can upload basic templates of possibly your own custom panels, and set x leds/panel, and a way to detect what pattern the panels are in.

I dont know how this would be done the easiest, and if you would have a cheap micro controller on each panel, or a strong one controlling them all.
I have an idea of how this may be possible. You could have a cheap, very simple micro controller in each, that can connect to a WiFi network that the main controller can control, and send info, such as, which panels are connected together, and where, using a data pin, to send a number (1) from panel 1 (Main panel) to panel 2 (number 2), along with the panel side id.

Lets say i make octagon panels, and could do that. I would use 5 pin connectors to send power and all data, both for leds and and the location detection, (1 data for leds, 2 for location detecting (1 for sending side and panel id, and 1 for receiving side and panel id) 1 for + and 1 for -)
My panels' sides would be 15 cm long (inside), which is 4 leds/side (If you use 30 led/meter led strips), aka 32 leds/panel. Now, then i would be able to set an animation, that preferably would be made for panels specifically, so transitions would be animated seamlessly. One way to do this, would be have a software generate animations for you, either from a video or a gif, of by using rng, to create some simple effects.

The YouTuber, Nerdforge did something similar to this whole led panel thing ([here](https://www.youtube.com/watch?v=2OP-oEbzB6g)), and it's close to what i would love to see in WLED, but just not quite enough. 

I would love to help, if it is something that may become a feature at some point, to the extend of my knowledge of arduino coding, and micro controllers.

MajorLOLGaming, 2019-11-11:
I have made a few prototype pcb's for it, if it becomes a thing.
This is the main board to control everything. It is an ESP-32 (ESP-WROOM-32), that is meant to control the leds/animations, and correctly setup all panels, using info from the subpanels, and magnetic reed switches (8/panel) which tell it which panel sides are against each other.
[image]

This is the sub boards, which will in all other panels (sub panels). It has an ESP-12E (ESP-8266MOD), which is in charge of sending panel info, like panel sides connected with each other, using magnetic reed switches, in each side of each panel, (8 each) and then using that information to calculate the position of the panel. 
[image]

raoulteeuwen, 2019-11-11:
Hi @MajorLOLGaming . Maybe this is something different, if so, sorry, but when someone talks about custom PCB and LEDs, that makes me think of https://quinled.info/2018/09/08/quinled-why-and-about-the-history/ ... Maybe you want to check it out.

MajorLOLGaming, 2019-11-11:
@raoulteeuwen It seems that what you have linked, is much different, but still quite cool.

Aircoookie, 2019-11-12:
@MajorLOLGaming wow, you are really enthusiatic about this! Those PCB drafts don't look bad!
But how can we see that WiFi microcontrollers have gotten too cheap? : D
You are considering putting an ESP8266 in every single cell, that is a bit wasteful.
Because one ESP (even 8266) can easily drive up to a thousand LEDs, just using some LED strips in the panels and daisy-chaining all of them will work out just as well.

As for software support, two upcoming WLED features are/will be very useful to achieve this sort of thing without any further software modifications!
- Segments! They enable you to specify ranges of LEDs (so you can have a segment per panel) and set those segments to different colors or effects each! The entire configuration will then also be store-able in a preset and can be restored with one click.
- LED grouping. This essentially will allow you to make a few consecutive LEDs act as one, which could be useful if you want to create the effect of one panel/tile only being a single solid color, even though it contains multiple LEDs.

What do you think?

jerryngm, 2019-11-13:
@MajorLOLGaming I think what you're trying to do is very similar to this [https://www.kickstarter.com/projects/photonlab/element-light-the-worlds-smartest-and-most-modular](https://www.kickstarter.com/projects/photonlab/element-light-the-worlds-smartest-and-most-modular)

They use an ESP32 on each of the panels, however rather than using physical connections they use WIFI Mesh and an app to define orientation. 

I have received those panels and loaded WLED firmware on it. WLED is the best wireless LED stripe driver for ESP boards🥇

MajorLOLGaming, 2019-11-16:
@Aircoookie thanks for the compliment :) 
I know the 8266 is overkill, but at the time, I couldn't find anything else on EasyEDA. 
The upcoming features sounds very promising, and would seem to be the right thing to use here. Thank you very much for your time and help.

@jerryngm dang, those are cool. While that would be awesome, it's a bit expensive. But of course if you compare it to nanoleaf, it's a complete steal. Both because of led layout, and the price. It is nearly what I'm imagining, just a lot more... Modular. This could be cool to take a look at, maybe.

closed 2020-04-05 by stale[bot]