wled/WLED #341: [Feature Request] Enhanced Schedule

mpmullally, 2019-11-11:
I was thinking about a way to expand the current timing functions available. A couple of options that would make it easier to set and forget. 

I don't know what would be the best but a couple of ideas: 
- Google Calendar integation where the title of the event or the description contains one of your macros and it runs from the start of the event until the end and at the end it turns off. 
- Calendar/agenda in the app that allows you to add items with start and end times again using the macro api.
- Add integration with a Sunrise/Sunset API call to schedule things based on Sunset at your location. (i  think most people would only want to turn on at sunset and turn off at a set time)

I have beginner experience with JSON APIs but would enjoy more information on how to use it better, does it allow for scheduling? If the easier way is for me to write out a JSON file with all the information i need for the next month or so i could do it.

Aircoookie, 2019-11-12:
I really like your suggestions, and I want to at least integrate the sunrise/set part of it natively at some point (likely OpenWeatherMap) as I would imagine a lot of users wanting to turn on the lights at sunset! (of course my point below also applies to this, you can easily do this using HA)

About the calendar support, that seems a bit advanced to implement, but I'll keep it in mind. Creating advanced schedules via JSON is easier, but scheduling features like this have excellent support on various home automation platforms (HomeAssistant is currently getting native WLED integration!), so it might be a better idea to use that for now :)

sujitrp, 2019-11-20:
Aircoookie Please have look at this implementation @ timers and NTP based time.
Some functionality is already in WLED.
https://github.com/arendst/Tasmota/wiki/Timers

Aircoookie, 2019-11-20:
@sujitrp Seems like they pretty much have the same feature set WLED does, apart from the random time offset and sunset/sunrise calculations. The latter is something I'd really like to add to WLED and it's nice that they found a way to calculate that from lat+lon so I don't necessarily need to use a weather API straight away!

garyd9, 2020-08-09:
A usermod was recently added that uses Dusk2Dawn to calculate sunrise and sunset times, and then runs a macro at those times.  At the moment, it's not a very elegant solution, but it does serve the purpose of "do something at sunrise" and "do something else at sunset."

Using the usermod requires a bit of code editing (to set your latitude/longitude, any offsets desired, and to confirm which macros are run at which time..)

cssinate, 2020-10-27:
I just wanted to suggest that if this gets looked at again that the Enhanced Schedule aspect should probably implement iCal RRULEs, both for implementing/scheduling in the interface, but also being able to parse external iCal files. Benefits of this approach:

1. Named schedules (i.e. "Christmas Season", "July Fourth Weekend", "Halloween")
2. Could use the Summary for macro data
3. Easier access for usermods to implement schedules created/managed by Google Calendar/Outlook, etc.
4. Options familiar to anyone that uses the above applications
5. There's probably an open source library that already implements most of the logic

blazoncek, 2021-10-03:
Sunrise/sunset triggered presets available since 0.12. Closing.

closed 2021-10-03 by blazoncek