wled/WLED #342: led stuck on a color

alfredolo, 2019-11-11:
Hello and congrats for this wonderful firmware.
I have some working led strips. After my daughter played fast on settings one of the led on the strip got a led stuck on a single color (green becouse there was a green effect). After that no changes to fix it, only power down the strip and wemos worked.

Do you have any idea why?

I am using :

wemos d1 mini pro v2
power shield for 12v power
74ahct125 as level shifter
1000uf connected to power suply 12v
470 ohm resistor close to led strips
ws2815b 12v

Thank you

Aircoookie, 2019-11-11:
Interesting, I have no idea as to why this could happen... Did the issue go away after powering back up? 
Maybe a bit in the output pixel buffer became stuck, but i never heard of that actually happening.

alfredolo, 2019-11-11:
Thank you for reply.
Yes after turning off and back on it worked.
I have already seen this problem while doing assembly tests in recent days. The problem seems to be connected to a disturbance on the line that makes one or more LEDs go crazy and no longer respond to anything until the voltage is removed.
I tried several level shifters and now with the 74ahct125n which is considered one of the best with a 1000uf capacitor on the line I had no problems until now.
Perhaps a small disturbance on the signal, I try with a 0.1uf capacitor in parallel with the 74ahct125n power supply like someone suggested somewhere.

closed 2020-04-05 by stale[bot]