wled/WLED #343: White LED Night Light

dpackham, 2019-11-11:
can you add a popular led effect that I see around is a single-pixel white LED every few meters around the house?  think the old school can light that are in the eaves every 2 meters.    this would be a NON-holiday effect for all the other nights of the year.   maybe allow the num of pixels skipped to be configurable?

Example pic below

[image]

kcoop99, 2019-11-12:
+1 would like to see this.

lpacchin, 2019-11-12:
nice feature

Aircoookie, 2019-11-12:
Yes, i can easily add this effect soon!

ironosity, 2019-11-13:
I think it could be really cool to be able to just specify a pattern for off leds.  Thus you could specify 1 led on, 10 off with primary white to achieve OP's objective. But if you just have close LEDs and want to space them out, you could specify 1 on, 2 off (spacing out 8-9 inches) but then the effect, color options on would run as though ON LEDs are all that matter. So you could still chase, or flash, or whatever but it would treat ON LED as the sequence.

lpacchin, 2019-11-13:
Suggestion 

Use the primary color picker as usual
Use the speed slider to set the number of LED on
Use the intensity slider to set the number of LED off
Repeat sequence for the number of LED available in the strip

What do think?

dpackham, 2019-11-13:
could it just be

Use all current sliders as now implemented
Use a new SKIP box to set the number of LED skipped starting from first LED?

this way ANY effect would work?
save to a preset and boom..

brnielsen44, 2019-11-15:
I really like dpackham's solution as that would then also work with various other effects.  I also think there could/should be an effect that just alternates between your primary and secondary colors. That way, for Christmas (for example), you could have Red x x x White x x x R x x x W x x x R etc. Thoughts?

ironosity, 2019-11-15:
@brnielsen44 I agree this would be useful. A setting could be added to LED Setup called Repeating pattern. for simplicity it could be a comma separated list of hex colors. Off could be black for consistency. So `F00,000,000,000,FFF,000,000,000` would repeat `[Red,off,off,off,White,off,off,off]` over and over. 
Some additional options:
You could default the effect to just alternate primary and secondary with no space when no pattern specified in settings.
Effect speed could be `solid` for 0 and do some other effect such as `running 2`
Effect intensity could add off leds between pattern
Effect intensity could add repeating on (or off as well) leds such that above pattern would be `[r,r,x,x,x,w,w,x,x,x]`
These are just initial thoughts.

brnielsen44, 2019-11-15:
@ironosity Yes, I thought through that also... only reason I didn't suggest that as the first option is I kind of like the idea of skipping LEDs with some of the other effects also (like Circus or Halloween) but yes, that would work for various effects.

Also, noticing a lot of LED folks are from Utah. I just moved to Lehi from Texas (and just learned that Dr Zzs is from Alpine area too). Small world.

ironosity, 2019-11-15:
that's funny. Small world! shoot me a message! Let me know where in lehi you're at! https://www.facebook.com/ironosity

dpackham, 2019-11-15:
AF here...  we should have an LED meetup :)

ironosity, 2019-11-19:
Whoa! Incoming!? https://github.com/Aircoookie/WLED/commit/3aacb7150d7a1d67bed931650d22badbc982204d

kcoop99, 2019-11-19:
NICE! Ok.....so how do we add it?? Can it be added via OTA?

ironosity, 2019-11-19:
@kcoop99 It will be in next binary release presumably since it's in master. But if you want it now, you'd have to compile the latest master creating your own binary and flash that.

Aircoookie, 2019-11-19:
Yes, it is in! However it is near impossible to set the amounts of on and off LEDs precisely with the sliders, this mode is more meant for APIs. I will add another "spots" mode soon which will let you specify how many "spotlights" you want and how many LEDs each one should be :)

kcoop99, 2019-11-19:
Awesome! Can you direct me on how to update my current version?

lpacchin, 2019-11-19:
@Aircoookie 
It works very well, thank you!
If it possible to add a   +    and    -    buttons near the slider, it is easier to set the amount of on and off LEDs, a sort of fine tuning

dpackham, 2019-11-20:
Downloading now.

joelevi, 2019-11-20:
> AF here... we should have an LED meetup :)
> 

I live in Cache County and work in Clearfield. I'd love an invitation to this "Utah LED Makers Meetup". :)

ironosity, 2019-11-21:
Ok, I finally got code to compile and took some photos showing what @dpackham was asking for. Phone photos...in the dark...so not great. :) https://photos.app.goo.gl/9zBYbhGkTJ8iAstK7
IRL, it looks good. Thanks @Aircoookie!

kcoop99, 2019-11-21:
@ironosity that looks AWESOME! I am going to go out on a limb here and say that this effect is still not ready for average Joe right now? I'll need to wait until it is in a release from Aircookie. Hopefully the next realease has it!

ironosity, 2019-11-21:
@kcoop99 I haven't noticed any problems with it. If you don't want to go through the hassle of compiling, here's the one I compiled. https://drive.google.com/open?id=1YJDpRD8rb5ygJ-HR9H5NPXjoDH0LVzqQ

kcoop99, 2019-11-21:
@ironosity great work! I do have a few NodeMCU's around but would this .bin work on my D1 mini?

ironosity, 2019-11-21:
@kcoop99 oh...no. Sorry. That was my first compile so I didn't even think of that. It wouldn't work. It was for a esp2866 12E nodemcu board.

Aircoookie, 2019-11-21:
@ironosity @kcoop99 thanks for sending the binary, I need to make a PIO setup that auto-creates the binaries for me... About 12E, Wemos D1 mini also uses that chip, so the binary will likely work :)

debsahu, 2019-11-21:
@Aircoookie this is what you are looking for: https://community.platformio.org/t/how-to-deploy-a-release-with-travis-co/294

fijit586, 2019-12-04:
Absolutely. Great Idea!!! And I agree the color should be whatever color you want, not just white.

IN FACT, one could include "Random Static" and get the same effect as the multi-colored xmas lights on a a string. The colors are choosen at random when initialized and remain that color for the duration.

ironosity, 2019-12-04:
@fijit586 in the latest code, there is a new "static pattern" fx that uses prim, sec, tri colors. AND I added a setting to disable LEDs in order to spread out colors (this works with ALL effects).

fijit586, 2019-12-04:
Ah,
I’m running v8.5.
Thanks, i’ll check and reload wled.

Regards,
Fred Tangelder
248-953-5122

dpackham, 2019-12-04:
I don't see a "static pattern"  only a "solid pattern"  and no way to disable LEDs...  found it.  in settings Disable Repeating N LEDs  #  

so every time I go to select this pattern I need to enable this setting?  and when I want to go Christmas colors on all LEDs I need to disable this?

hmm maybe need a macro of quick set etc.

ironosity, 2019-12-04:
@dpackham yeah, Solid Pattern is what I mean. or Tri Color Pattern for 3 colors.  The disable N Led isn't specifically meant for this pattern. Rather, it's meant to just spread out on LEDs for all patterns/effects. Typically Christmas light strands space things out 5-9" where as LED strips/strings are much closer. Thus the setting is global. Everlights, a popular commercial permanent light installer spaces theirs out 9"

dpackham, 2019-12-04:
Right.  I would just like to see the Skip button on the main UI to toggle that effect of skipping N number of LEDs.   ill have to figure out how to use HomeAssistant to toggle I think with the API.  like Christmas lights on all LEDs to the solid pattern on every 6th etc.

kcoop99, 2019-12-05:
Can you please share how this can be implemented in my current version 8.6? Can it be updated OTA? Is there a simple process? Thank you all for your work on this!

ironosity, 2019-12-06:
@kcoop99 It can't be done in 8.6.  You have to use the binary I posted above or compile it yourself.  You can flash it by going to Settings, Security & Updates, Manual  OTA Update. select the .bin and press Update.

closed 2019-12-13 by Aircoookie

nstanoszek, 2021-09-30:
is this still available? I don't see "static" anything in effects.

sansillusion, 2021-10-01:
The effect name is Spots.