wled/WLED #346: power supply control

pauldogg, 2019-11-12:
Would it be possible to add delay in ms for power supply to come online, as well as a pin definition for power supply relay?

Thanks

Aircoookie, 2019-11-12:
See #345 for where the pin definitions are :)
May I ask what is your use case is for a delay in relay switching? For me, turning the relay on whenever LEDs are on and off when the LEDs are off is the only logical way.

pauldogg, 2019-11-13:
Sorry I meant delay the signal, allowing time for the power to get to the LEDs.
In my current setup I have it set to 20ms; on another, I need to set it to 1000ms. If i didn't have a delay, the lights would appear off until another signal was sent (change color, brightness or effect etc...).

Aircoookie, 2019-11-15:
Have you tested that that is an actual issue with APA102?
WLED sends an update to the LEDs at least twice a second.
If the LEDs actually stay off, please confirm whether my hunch that only changing brightness will make them work, changing effects or color should not have an effect.

Fejitatete, 2020-05-31:
Is it possbile to reopen this issue plz, I have the same need : I power my strip from an ATX PSU and when the power signal is grounded, actual power may take 2s to come. Without a delay, the fade in effect is lost. Another way is to wait for an input <PSU power ok>.
Thanks.

Fejitatete, 2020-06-06:
Thank you @Aircoookie, keep up the great work :).

Maurotb, 2021-04-11:
@Aircoookie i have download 0.12.0 , this enanchements (delay output after relay on) is not present?
Thanks

Sangoku, 2021-12-19:
Ye sadly that code snippet is not working. I corrected it in the #812 (comment)  and made it work. it breaks on color change.

blazoncek, 2022-08-18:
The delay in powering up can be achieved by using playlist feature introduced in 0.12 and enhanced in 0.13.
Create 2 presets, one with on state and black color, 2nd with the desired effect and color and transition time. From those two create a playlist and use it for powering WLED.
Power off delay has been implemented in code.

up2smoke, 2022-09-03:
where exactly in the code?  for the people who compile themselves, that would be interesting to know

closed 2022-10-10 by blazoncek

KlausHans, 2023-09-06:
Another problem with the playlist solution: If you use a potentiometer for brightness the solution doesn't work because the global brightness via potentiometer has nothing to do with presets. So a custom relais delay setting for switching on would be nice.

KlausHans, 2023-09-06:
@Aircoookie has mentioned this issue in https://github.com/Aircoookie/WLED/issues/812#issuecomment-640874076 as main tracking issue. So maybe reopen it since it isn't implemented yet and the workaround is only half baked (imho)?

doronazl, 2023-12-12:
> > If i change to json view i can set `"dur":[1]` (0.1 s) but that doesn't get saved. Instead the default value of 100 (10 s) is saved. This seems to be a bug or at least a big inconvenience.
> 
> Please open a new issue. But my own testing shows that it is possible to enter 0.1 as a preset duration.

tried entering anything lower than 1s on multirelay menu, saving it , then go back to the menu it set back to 1, how can set under 1s ?

KlausHans, 2023-12-12:
Did you try to save 0.1s? It has to be 0.2 or higher, as discussed in https://github.com/Aircoookie/WLED/issues/3359

doronazl, 2023-12-12:
> Did you try to save 0.1s? It has to be 0.2 or higher, as discussed in #3359

yes tried decimal point, it wont save

doronazl, 2023-12-12:
> #3359

that discussion is about the playlist, over there no problem it takes the decimal point, but not in usermods

KlausHans, 2023-12-12:
True. Sorry, misread it.

doronazl, 2023-12-12:
found this in the example file

     * Usermod Settings Overview:
     * - Numeric values are treated as floats in the browser.
     *   - If the numeric value entered into the browser contains a decimal point, it will be parsed as a C float
     *     before being returned to the Usermod.  The float data type has only 6-7 decimal digits of precision, and
     *     doubles are not supported, numbers will be rounded to the nearest float value when being parsed.
     *     The range accepted by the input field is +/- 1.175494351e-38 to +/- 3.402823466e+38.
but cant really make anything from it that helps me