wled/WLED #347: LittleVGL to C Array

jamesbahr, 2019-11-12:
Did little vgl remove support for converting gzip to carray? i remember doing it through that site but now it says it cant accept gzip files?

Aircoookie, 2019-11-12:
Yes, they added an annoying check to see if the input is actually an image file.
They are only checking the extension though, so you can still use the site by adding a .png or .bmp to the filename.
It's another step to take, I should really read up on Python to create a tool for simple CLI gzip + C array creation (+length) to make the process more convenient. Or I will ask if someone in the community would do it :)

jamesbahr, 2019-11-12:
its kinda crazy one doesn't exist yet in this IoT age. Thanks for the
advice on changing extension :)

jamesbahr, 2019-11-12:
https://littlevgl.com/image-converter/  and the site always goes down :/

jamesbahr, 2019-11-13:
i tried to install little vgl converter on a linux box as the site is down to see if i could get it working but its giving me same error?

~/lv_utils# php img_conv_core.php "name=mobile.c&img=mobile.png"
PHP Warning:  imagecreatefrompng(): 'mobile.png' is not a valid PNG file in /root/lv_utils/img_conv_core.php on line 66

Aircoookie, 2019-11-21:
Opened #357 for this :)

closed 2019-11-21 by Aircoookie