wled/WLED #348: Problem with preset numbers on iOS (13)

knochenkopp, 2019-11-13:
The numbers 9 to 16 in the preset-circles are recognized as a part of a phone number in iOS. They are blue and underlined. If I tap one preset in the row of 9 to 12, a window pops up on the bottom and say  "Call 9101112?". Same with the bottom row. If i tap a preset there the window pops up and say "Call 13141516"
The preset however will be executed.

[image]

Aircoookie, 2019-11-13:
Thanks for the report :)
Closing as duplicate of #262

closed 2019-11-13 by Aircoookie

knochenkopp, 2019-11-13:
Darn! I searched multiple times for a duplicate report, still missed #262. XD