wled/WLED #350: NodeMCU, Wled 0.8.6 & XLights

bcd19, 2019-11-16:
Hi there I'm new to using NoceMCU and Wled and I'm having issues with running a small test sequence in XLights.
The sequence will output to my 1metre strip of 60leds and be fine and in sync with music then at random points it will stay stuck on the set colours at that moment. Then after a while will resume its display further down the sequence. This happens everytime no matter what I do in Xlights just the point it happens varies each time, some times it runs longer others shorter but always freezes while the pc keeps going through the sequence.
These leds are wired to a 5v 30a power supply which also powers the nodemcu from seperate output to lights. The data wire to first led is about 6" long but when it is working for that bit its displaying perfectly fine and looks great.

Please any help or advice is greatly appreciated,
Cheers

marc-gist, 2019-11-16:
@bcd19 how far away is the NodeMCU from your WiFi router?  Could be a network communication issue i would guess.

bcd19, 2019-11-16:
My wifi router is a little over 20feet away but it is quite congested with
lots of other devices connected.

bcd19, 2019-11-18:
Update :
So I have grabbed a spare wifi router not on internet and connecting just my desktop and the nodemcu.
The router is sitting next to the nodemcu and the desktop is wired into the router by ethernet. I have checked the voltage to leds with multimeter and i'm getting around 3-4amps depending on effect.
Everything still does the same as my first post the lights will display my sequence for abit then freeze on the last color before resuming at some point later in the piece. This is driving me nuts and I have done everything I could think of to fix this.
Could it be a faulty nodemcu? as it does work but over say a 1min test sequence may freeze and resume 3-4times.

Thanks again

raoulteeuwen, 2019-11-18:
Could you check whether https://github.com/Aircoookie/WLED/issues/255 helps?

bcd19, 2019-11-19:
Raoul thankyou for linking #255 disabling NTP server worked for me and now there is no issues with my sequence. 
Thanks for the help

closed 2019-11-19 by bcd19

betadeltadelta, 2019-11-19:
Excellent find!

I too had the same issue with freezing consistently every few seconds.  It turns out I've placed several 8266's running WLED on a separate wireless network, not connected to my home network, nor the internet.  They were all trying to reach out to an NTP for the time.  

This brings up a question, I placed them all on a dedicated network, not connected to the internet, to allow for faster more consistent E1.31 via Xlights.  When Xlights is not running, I intended to turn them on and off and run WLED presets with macros.  Xlights merely overrules whatever WLED is doing/commanding via UDP, and then when Xlights stops, the MCU's revert back to what WLED is commanding.  Works great in theory.

But, it turns out you cannot do so without the proper time, obviously.  Is there another means anyone knows of to set and forget the time, without using an NTP server?  Perhaps this is implemented to reach out to a server since ESP8266's have no clock battery or means to retain the time, so at power off they'd have to reach out every time for the proper time at each reboot/powerup?

Aircoookie, 2019-11-19:
@betadeltadelta yes, you can set the time both via the HTTP and JSON APIs (commands are `win&ST=u` and `{"time":u}` respectively), where `u` is the current Unix timestamp!

betadeltadelta, 2019-11-19:
Of course!!! Awesome.  I thought I recalled seeing it somewhere in the wiki but couldn't track it down.  Thanks!  I guess that syncing via UDP is only for LED settings, correct?  So if I use this command via http to set the time on one Node, it then will not sync to all other nodes when the sync slider is selected. I'll have to test but only have one NodeMCU with me at the moment.  Thanks!!!!

Aircoookie, 2019-11-19:
@betadeltadelta yes, that is correct, it will not do that currently. That said, it is painless to incorporate the time data into the sync packet, so I'll consider adding that! Would also mean that only one ESP in the network would need to connect to the NTP server!

betadeltadelta, 2019-11-19:
That was kind of my thinking, like another check box to sync time to others as master/slave setup.  99% of your users probably use NTP server checked and I would too, but my home network is so busy with gaming consoles streaming video devices, alarm system and the lot that it makes all of the ESP8266's lag or freeze from time to time.  Therefore, I put them on their own cheap TP-Link router that is not connected to the internet.  Then I realized I cannot use macros to turn them on and begin sequences after dark outside my home, because they lose the current time.  Currently looking for a workaround.  Considering still using this second router, but as a subnet fed from my main just to have internet/NTP server access, nothing more.

Love love love the project!!!  Found this about a month ago from Dr. Zzzs and watched your webcast with him.  Excellent project!