wled/WLED #351: Sync/timing Between multiple

39mbac, 2019-11-16:
How can you control/sync timing between multiple WLED deployments - example I have 2 WLED systems deployed on 2 separate nodemcu working great! The one issue I am running into is keeping them in real sync with each other - ie running wipe(3) effect  on WLED 1, WLED 2 will trail just a bit behind 1 - the sync works fine. I would like them to run the wipe identically or any animation actually.

Thanks for this incredible work. I am a novice at this and really appreciate all of your effort!

brentbrooks70, 2019-11-16:
@39mbac The first thing I would try is to set your device to use NTP (_Settings -> Time & Macros -> Get time from NTP server:_ **is checked**), I think some of the newer timing methods are tied to the time so this might help....Not positive about that but couldn't hurt to try. 

Are the strips you are controlling the same length? What version of WLED are you running?

The only true methods to keep them in complete sync all the time would be to wire them to the same controller in parallel, or if in serial you could always "separate" them again with segments (not fully implemented yet). Or, use another device to provide one of the [real-time protocols](https://github.com/Aircoookie/WLED/wiki/E1.31-&-UDP-Realtime-Control) when you want them synced. 
Maybe someone else will chime in with additional info, in the meantime good luck

39mbac, 2019-11-16:
@brentbrooks70 I do have NTP enabled and strips are identical running 0.8.6 currently. I will check out the other info recommended and keep digging. Thanks!

closed 2020-04-05 by stale[bot]