wled/WLED #352: Sk6812 white channel switched with red channel

rybitski, 2019-11-17:
I'm using a wemos D1 mini running WLED 0.8.6 (build 1910255). Everything works as expected except the the white LEDs light up red and vise versa. Green and blue function as expected. 

On a separate note this is an amazing project, thank you for sharing.

Def3nder, 2019-11-24:
Hi @rybitski ,

what type of Color Order did you choose on the LED Preferences page ?
```
> LEDs are 4-channel type (RGBW): [ ]
> Color order: [GRB]
```

The RGB Version of the SK6812 has color order [GRB] while the RGBW version has color order [RGB].
So, you need to enable RGBW and change the color order:
```
LEDs are 4-channel type (RGBW): [x]
Color order: [RGB]
```

Does this change to the expected behaviour ?

rybitski, 2019-11-25:
This was an issue of using 3.3v logic. It worked on the bench, bit when I installed it I experienced the issue. Adding a logic level converter solved my issue. Thank you for the response and sorry for the confusion.

closed 2019-11-25 by rybitski

myhrmans, 2020-03-28:
I still face this issue after running this configuration:

```
LEDs are 4-channel type (RGBW): [x]
Color order: [RGB]
```
When running this:

```
LEDs are 4-channel type (RGBW): [x]
Color order: [GRB]
```
All colors are in the right order except white and red are still swapped.

The right configuration is GWRB.

rybitski, 2020-03-28:
What hardware are you using?

myhrmans, 2020-03-28:
> What hardware are you using?

https://www.adafruit.com/product/2847

rybitski, 2020-03-28:
My issue was solved by using a logic level converter. The esp 8266 chips operate on 3.3v logic, but the neopixels need a 5v signal. Are you using a level shifter?

mit11111, 2024-05-21:
Helllo We are facing an issue with sp105e controller while integrating sk6812_rgbw. Can anyone help with this please

mit11111, 2024-05-21:
> My issue was solved by using a logic level converter. The esp 8266 chips operate on 3.3v logic, but the neopixels need a 5v signal. Are you using a level shifter?

Can you help with sk6812_rgwb for sp105e controller