wled/WLED #355: Using Wled with Xlights - Universes

tinseltunes, 2019-11-17:
Hi. Firstly, Loving the software. I'm using it to control my arches both through the App and Xlights.

I'm also using it for my Megatree, and here's where the problem is.
I have 8 x 150node WS2812b strips on my tree which has 1200 nodes in total running off a D1 mini on the D4 pin. With a level shifter.
When using the app all the strips work perfectly, all 1200 pixels. Colours are good (lots of power injection), all the effects are working. I could run the tree just off the app.
BUT the effects in WLed don't really lend themselves to being on a megatree (no swirl or fan type effects.
I've set E131 and the universe number in Wled.

So I'm controlling the megatree through xlights and I've selected the universe number and I  have the universe size set to 510 in xlights. Only about 750 nodes will lightup using Xlights. 

Any ideas as to what's going on please?
Thanks

CaffeinatedFunctionality, 2019-11-17:
I had this issue. WLED is excellent but for e1.31 I got them all to light but they were badly delayed for anything over 700 lights. I'm not sure if it is a memory thing or something else. Next year I would be interested in looking at what espixelstick does differently for e1.31 as I can expand multiple sticks and have thousands of lights on that. WLED wouldn't expand past 2 nodemcus. I love WLED and would rather use that so I hope a solution is found. Just be aware others have similar issues and if using e1.31 you may want to use espixelstick until a fix is found. Though if you somehow do get this working let us all know. Maybe I am doing something dumb. I use them all unicast on a network with nothing but FPP and the nodemcus so it shouldn't be network congestion. 

⁣Sent from BlueMail ​

Aircoookie, 2019-11-18:
Hi! I just updated WLED to the ESPAsyncE131 library. It is now be possible to address up to 1190 LEDs (sorry tinseltunes :D that's the limit for 7 universes, just 10 LEDs short) and it should now work better than before, at least with the 50ms xlights mode! The configurations I tested are 500 LEDs at 60fps and 1000 LEDs at 30fps.

tinseltunes, 2019-11-18:
Great thank you for the replay and update to accommodate this.
Shame about the last 10 LEDS. 
In the future I would love to move away from Xlights. It's just the megatree effects that I need.

tinseltunes, 2019-11-19:
How come there can't be 8 universes?

Aircoookie, 2019-11-19:
@tinseltunes there can be. I decided to draw the line at 7 because that was the maximum that seems to be working with a decent framerate. For a slow moving megatree effect where 20 fps are enough though, up to 10 universes may work!
You can change it in line 440 of wled00: `#define E131_MAX_UNIVERSE_COUNT 8`.
I might even consider setting 8 as the default limit with a warning recommending to not use more than 1000 LEDs (which is pretty much my recommended limit for how many LEDs you should run with WLED anyways)

tinseltunes, 2019-11-20:
Awesome. Do I edit the file then upload with arduino ide? As I tried this (loaded the libraries) and got all sorts of compiling errors not related to the libraries. Originally I uploaded the bin file with home flasher.

illuxions, 2019-11-23:
tinseltunes,   look at the errors in the compiler, duplicate library warnings are ok, look for the libraries you are missing.  Google them then download the library and unzip it to your arduino libraries folder(keeping the folder name), or clone it to keep up with the source.

Aircoookie,
I think I'm seeing the same issue with 300 LEDs on a NodeMCU 12e.  

Just a wild guess, but it seems a single multi universe send isn't getting handled correctly. I built with the latest code and also tried all the way back to .81 binaries.  

If I address universe 1 or 2 by themselves it's fine, but addressing both universes it pukes.  I'm not that adept with Arduino, but I'm trying to figure out how to throw some stuff to the console so I can help to debug it(perl and c based old fart trying to get my feet wet)

I tried Xlights 2019.65(and .64), sACNview, and da_e131 as test software.  I tried controller type of unknown and epixelstick.  I moved those same strings to a PI and used Falcon Pi Player as an E1.31 host and it was ok with the 3 software tests addressing both universes. A beaglebone running FPP with an 8 string F8-B cape has no issues with the 3 softwares either. With WLED I can get either universe to wiggle discreetly but can't get them both to work at the same time.

I've been digging at it and I'm at a loss.  I'll try it on an ESP32 tomorrow to see if there's any difference.

Aircoookie, 2019-11-23:
@illuxions thanks for the feedback! What exactly do you mean by "it pukes"? Does WLED crash or become unresponsive, is the framerate way too low or are the colors off after about 170 LEDs?
In the last case, the sender software needs to send all three RGB bytes of a pixel in the same universe. So the maximum universe length is 510 (512 is the standard). In LedFx (the sound visualization program I use for testing) you can set that. If some important software (like for example Xlights) doesn't support that, I'll have to change the logic but that will slow down the implementation a bit.

By the way, did you try Unicast, multicast or both?

illuxions, 2019-11-23:
@Aircoookie  No problem, I'm happy to a buy a beer or 4 or people sharing good toys that I use. 
 
Unicast is all I can try for now.  My main router doesn't support IGMP Proxy and Snooping, I'll dig for one of my old routers so I can flash DD-WRT on it to try multicast.  

I tried frame delays in da_e131 but when there's more than 1 packet with a different universe nothing happens at all, wled behaves as if it doesn't see the packet.  This is the same behaviour I see in Xlights as well and using the beaglebone running FPP and an e1.31 output. I also tried setting different universes as different outputs with different priority on the beaglebone with the same thing, wled just sits there.  It doesn't lock up, I have 2 instances of da_e131 open, 1 set to universe 3-3 and the other set to 1-2.  3 fires ok, but 1-2 does nothing.

Watching with wireshark, it seems that when there is more than 1 universe all of the software sends them in consecutive packets with the universe ID in each packet.  

I wonder if it's something with the E1.31 ring buffer.  I'll trying to play with that to see what I can break

Here's just part of the madness I'm playing with. [https://i.imgur.com/gqTUmdP.jpg](url)

illuxions, 2019-11-24:
I'm sitting in a pile of routers trying to nail down this issue, it looks like the router must support IGMP or even unicast to multi universes doesn't work with the fireineye.  I spent 8am - 7pm so far toying with this.

Test setup:
NodeMCU ESP12e with WLED .87-dev master code with 4 x 50 WS2811 on MainNet
NodeMCU ESP32 with WLED .87-dev master code with 4 x 150 WS2812 on LightNet
FPP Master in Docker on Synology NAS on 'MainNet ' - FPPM
FPP Remote on Beaglebone black on 'LightNet' - BBB
Linksys WRT3200ACM(no IGMP - stock latest firmware) - MainNet
TPLink Archer C3200(all IGMP features - stock latest firmware ) - LightNet

Also tested with:
Asus RT AC68U (Asuswrt-Merlin firmware updated to latest 384.13_1) 

Testing results for Multiple Universes;
FPPM -> ESP12E -> Unicast No, no multicast on router = no multicast
FPPM -> ESP32 -> Unicast YES, Multicast can't traverse subnets without explicit IGMP proxy
BBB -> ESP12e -> Unicast No, can't multicast to MainNet
BBB -> ESP32 -> Unicast YES, Multicast YES

Single universe works Unicast in all tests.

The one that really caught me was the FPPM to ESP32, from a non IGMP subnet to the IGMP LightNet Unicast works!

I crashed my whole home automation, cameras, and other junk replacing the WRT3200ACM with the TPLink Archer then fired up the AC68U so both subnets have full IGMP.  Multicast works on each subnet and Unicast works across subnets or their own subnet with WLED.

I don't know IGMP networking interaction so I'll abuse one of my networking guys on Monday to see if we can understand why.  

Unicast _should_ work without IGMP, it works with the other hosts like FPP as an E1.31 bridge with a Pi or BBB. Maybe it's something in the ESPAsync libraries.  

Tomorrow I'm going to flash my WRT3200ACM with DD-WRT and see if that makes it happy.  I'm also going to peruse the FPP code and compare it with the ESPAsync libraries to see if I can't find anything useful.

pierslawson, 2019-11-24:
Hi to add in, I was using the 0.8.5 Release Binary with 625 LEDs in 4 universes from XLights. The communication was sort of working, i.e. the lights would go off as I told XLights to output to the lights, then the display would start quite a long time later with a lot of "holes".

I've just built the latest code for NodeMCU V2 and tried that. Again the lights turn off when I tell XLights to output to the lights, but now there is no display at all.

tinseltunes, 2019-11-24:
Aircookie, are you able to release a bin file that has the 8 universes please? I'm having so many issues trying to compile the wled000. Ini in arduino ide. That would make my life so much easier. Please please. 😀

pierslawson, 2019-11-24:
@tinseltunes you could try using VSCode with the PlaformIO plugin. This worked for me #374 but as I said above, there were still problems with the ESP8266 and e1.31

tinseltunes, 2019-11-24:
I'm not sure what that means. Sorry

pierslawson, 2019-11-24:
VSCode = Visual Studio Code, a free IDE fro Microsoft. PlatformIO IDE is one of the plugins available for it.

pierslawson, 2019-11-24:
> I've just built the latest code for NodeMCU V2 and tried that. Again the lights turn off when I tell XLights to output to the lights, but now there is no display at all.

My bad... I had the universe IDs set wrong. So I can confirm that when there is a single universe XLights can control WLED and it is responsive. However, if I change Xlights to use 2 or more universes WLED stops responding altogether.

pierslawson, 2019-11-24:
> However, if I change Xlights to use 2 or more universes WLED stops responding altogether (this is with a build of the latest master).

My bad again... I think I have been misunderstanding XLights. I now have for universes to control 625 WS2812B  LEDs and it is working fine. The response time between starting a sequence in XLights and it appearing is very good. I am using Unicast.

tinseltunes, 2019-11-25:
@pierslawson did you manage to get your's working. I still can't drive more than one universe without the other lighting random pixels. I've tried to compile the Wled000.ini in Visual Studio and change the universe number to 8 but VS can't fix all the #includes errors. 
I'm in so much trouble trying to get this to work with xlights patterns on my mega tree.

Aircoookie, 2019-11-26:
@tinseltunes sorry for all the trouble :( I meant to send you a binary supporting 8 universes but forgot about it... Will send in about 8 hours!

tinseltunes, 2019-11-26:
That would be awesome. I appreciate all that you do for this community. Someone said something in a comment about buying you a coffee or something? Is there a donate system?

thedannymullen, 2019-11-27:
> In the last case, the sender software needs to send all three RGB bytes of a pixel in the same universe. So the maximum universe length is 510 (512 is the standard). In LedFx (the sound [...]

@Aircoookie xlights supports changing the universe size.  I use 510 for my universe size.

tinseltunes, 2019-11-27:
Hi, Sorry I was just seeing how you're getting on with that bin file. My display is due to go live on the 1st December. I appreciate your help.

illuxions, 2019-11-27:
@tinseltunes 

I'll clean this up later and maybe it can be a wiki page.

For a window install;

I uninstalled arduino and deleted my AppData\Local\Adruino15 folder.

For something using a CPO2102 USB UART these are the drivers drivers https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers.  If you are using a bare ESP8266 and an FTDI board, you'll need to find the drivers till your device shows up.

Using the zip file version I unzipped it to c:\Arduino meaning arduino.exe is in the root of that folder.

I ran Arduino.exe and went to preferecnces (crtl + comma) then I added https://arduino.esp8266.com/stable/package_esp8266com_index.json to the additional board manager urls, add a comma and https://dl.espressif.com/dl/package_esp32_index.json if you need ESP32 also.

Close preferences

Go to tools ->  board -> boards manager.

Type ESP and hit enter.

for ESP32 change the dropdown to 1.0.2 then hit install.  For ESP8266 hit 2.5.2 and press install.

I'm not sure if git or downloading zips it easier.  for now I'll show the git way.

I opened a command prompt and cd \, md 1, then cd 1.

Git clone https://github.com/Aircoookie/WLED.git

In Arduino IDE file -> C:\1\WLED\wled00\wled00.ino

Tools -> boards -> NodeMCU 12e for me changed upload speed to 921k, IWP 1.4 higher bandwidth, erase flash all, and port to com 3 which my NodeMCU is showing up on.  

You can look in the platformio.ini to get some hints on what's next
lib_deps_external =
  FastLED@3.3.2
  NeoPixelBus@2.5.1
  ESPAsyncTCP@1.2.0
  ESPAsyncUDP@697c75a025
  AsyncTCP@1.0.3
  Esp Async WebServer@1.2.0
  IRremoteESP8266@2.5.5

multiple library warnings  don't matter the important one it "NeoPixelBrightnessBus.h: No such file or directory"

Google "NeoPixelBrightnessBus.h" and find https://github.com/Makuna/NeoPixelBus

in the command prompt again.  cd \, cd arduino, cd libraries.

git clone --branch 2.5.1 https://github.com/Makuna/NeoPixelBus.git

I don't want to spend all night going through each error to show how to figure it out, so let's follow the includes from the platformio.ini.

git clone --branch 3.3.2 https://github.com/FastLED/FastLED.git
git clone https://github.com/me-no-dev/ESPAsyncTCP.git
git clone https://github.com/me-no-dev/ESPAsyncUDP.git
git clone https://github.com/me-no-dev/AsyncTCP.git
git clone https://github.com/me-no-dev/ESPAsyncWebServer.git
git clone --branch v2.5.5 https://github.com/crankyoldgit/IRremoteESP8266.git

(me-no-dev doesn't like tagging for some reason)

Go back to Arduino IDE...It builds. Do reset + program button routine on Nodemcu then press Upload in Arduino IDE.  Magic.  Open my phone and look for the wifi, feed it my router info(I actually changed it in wled00.ino so I didn't have to do that part) then change its name and start playing.

I'll walk through using VS Code, installing PlatformIO plugin, then letting it handle the dependencies this weekend.  I used Arduino IDE and google when I got started.

None of this will help if you can't traverse out of a single Universe.  It doesn't matter how long your strings are, you are addressing a specific set of pixels.  Are you using 12v strings?  Do you have have a buffer on the control line like a Hex inverting Schmitt trigger or other level fast level shifter?  With 12v strings the 3.3v out of the ESP8266 is just bordering not enough voltage.  It would be a lot happier at 5v.  You might be having the same router issues I was having as well.  If you can get 170 to work flawlessly but not 171, you need multicast but it's up to the router how it handles it.

Aircoookie, 2019-11-27:
@tinseltunes wow I am a dork, so sorry for making you wait! I did already compile the binary but forgot to send it 🤦‍♂ I just set a reminder so I'll definitely send it when I get back to my main computer in about 5hrs.

Aircoookie, 2019-11-27:
@tinseltunes Here you go, finally :) https://drive.google.com/open?id=1X5tZCVWzHSNnzgjLm-mJNnq5eCg3a_xK

Actually forgot about it again but luckily remembered before going to sleep!

tinseltunes, 2019-11-28:
I'm writing this in caps as I want everyone to know:
@Aircoookie YOU MY FRIEND ARE A GENIUS. The extra universe worked a treat, My megatree works like a dream. I can either run it with Wled or xlights with the full 1200 pixels. I'll pop up a link to a video in the next couple of days.
Thank you very much. I'll through a little $$ your way after payday. :)

pierslawson, 2019-11-30:
Hey, sorry for the delay in replying @tinseltunes Great you got there in the end... I only have 675 LEDs this year (so far) so hadn't hit the universes limit. @Aircoookie I can confirm the build I made from master is working very well with E1.31.

tinseltunes, 2019-11-30:
Looks great.. Here's a video.

tinseltunes, 2019-11-30:
Here's the display 90% finished. Thanks to everyone who helped especially @Aircoookie

tinseltunes, 2019-11-30:
https://youtu.be/iBllIS7s8M4

dhpete, 2019-12-02:
> Here's the display 90% finished. Thanks to everyone who helped especially @Aircoookie https://youtu.be/iBllIS7s8M4

video unavailable :(

tinseltunes, 2019-12-02:
Sorry, I took that one down. It only uploaded in 480p. New link. https://youtu.be/PmEHZLp7TSQ

thedannymullen, 2019-12-08:
Looks great!  Have any issues when xlights is idle?  
My display kept turning orange and I could not get that disabled.

tinseltunes, 2019-12-08:
Yes same thing happens. It only happens when xlights or x scheduler stops outputting.
So at the end of the night the house is off (not on wled) but fence, megatree and star stay lit until the timer on the power kicks it. 

I have turn lights off after powerout/reset too.

samstreak, 2019-12-09:
> My display kept turning orange and I could not get that disabled.

orange is the default color... go into the LED Preferences page and set whatever color/pattern/palette you want as the default on power up / reboot / no xlights(etc)

thedannymullen, 2019-12-10:
@samstreak I tried that but the setting would either not save or when they didn’t e1.31 did not function.  I plan on setting up a test rig after holidays to debug.  For now I switched to espixel stick.  
Their newest version had some issues also.  I suspect a library problem somewhere as common libraries are used.  I suspect This because when I build the binary with source from a year ago it  behaves the same as their current binary.  When I use the binary built a year ago it works.

PaulNBerlin, 2020-01-03:
Hi @tinseltunes,
I also want to build such a fancy pixel tree. 
Currently, I am struggeling with the cable management (see my professional picture).
You wrote, that you are just using one D1 mini. 
How, did you connect the strips ot the bottom of the tree ? 
How much spacing is there between the different strings?
How often did you inject power?

[image]

Thanks a lot for helping!

PaulNBerlin, 2020-01-03:
Just hat the idea using junper wire for longer distances at the bottom? 
Like this: http://s.click.aliexpress.com/e/buqEtUlE

raoulteeuwen, 2020-01-03:
Hi @PaulNBerlin . Why not make a led strip for each 'pole', and feed both power and data via a spliter to every (smaller) led strip?

PaulNBerlin, 2020-01-03:
Hi @raoulteeuwen, please correct me if I am wrong, but following your suggestion will result in same effects in each strip. Something like circles or moving text will not be possible.

raoulteeuwen, 2020-01-03:
I would not know, but didn't know what effect you were after. Maybe check out https://www.youtube.com/watch?v=ZFO_QOBG9Bs and https://www.youtube.com/watch?v=8a6JCIeZ6rU ?

samstreak, 2020-01-03:
I built a couple of small trees with a D1 Mini and 100 LEDs...  each vertical segment is essentially 12 LEDs high from bottom of the tree to top, and then between each down-to-up is 2 LEDs (which can be used or not... just have to set up your model)...  the building pattern is 2 LEDs at the start, then 12 to the top, 11 back down (since on my tree the 12th one is where the LEDs sit on the upper ring (see drawing below), then 2 LEDs... do that 2x then join the 2 50-LED strands and continue with the other 2 sets.  So there are 8 vertical strands.  I didn't inject power for the 100 LEDs, but I usually set my WLED to 35 brightness (out of 255) which is plenty for what I'm doing. If you would like, I could upload my small tree model I built for xLights.  Please note in the drawing below, the spacing on the actual physical trees are more even, but it would be hard to show on a drawing.

[image]

If you were to make it bigger (like 50-LED strand for each vertical), you could inject power from just the bottom, as power can go either direction through the strand, just the signal wire needs to be continuous from start to finish.  

Hope this helps!

PaulNBerlin, 2020-01-03:
Hi  @samstreak,

thank you very much. You really gave me a lot of extrem valuable input!
I am quite eager seeing your XLights model (would assume the config for shutting down the 2 leds is made there!?).

May I ask another question: If I using 16 string with 50 LEDs each, I need to think about power injection. I would use a 20 Amps power supply (to much?). Are there any consideration required regarding AWG cable strenghts (e.g. 18g or 20g or something totally different).

Thanks and best regards,
Paul

samstreak, 2020-01-04:
I haven't figured out yet how to turn off those LEDs in xLights, just in my WLED controller (with the 0.90b1 you can set up segments, so I did that and WLED is much better for these trees)  for individual autoruns without the e1.31 running...  but here's the model I've got right now. I haven't gone over it with a fine-toothed comb yet to ensure everything lines up 100%, but it's close enough for my first attempt at learning this stuff.

[Tree1.zip](https://github.com/Aircoookie/WLED/files/4022417/Tree1.zip)

samstreak, 2020-01-04:
I think I just found A way to do it - not sure if it's the "right" way or not...

In the tree model, go down to the 'States' in the first section (mine is Poly Line because that was what I based my tree from), click the --Click To Edit--, then click the "..." box to bring up the States Definition box.

Click the Add button to add a new "state". I called mine SkipBottom, but whatever makes sense to you...

Check the Force Custom Colors box

In the first row of the list box, give the State column a value (I just put a 0, but 'off' or whatever makes sense to you is good)

In the first row Nodes column, click the empty box and hold control button on the keyboard while you click on the nodes you want to disable. Please note that WLED and xLights differ between the LED notations, so WLED's first LED is 0, while xLights' first LED is 1, so the numbers in my previous post won't work right on xLights - you need to add 1 to all of their numbers.

In the first row Color column, click it and pick black as the color. 

Click OK to the State Definition box.

Go to the Sequencer tab, and insert a new "State" effect on the model (box with a black A on a white background and a white B on a black background), and pick the state name you made earlier in the State Definition box. Then in the States group box, with the State radio button selected, pick the 0 (or 'off' or whatever you put in the first line of the States column of the State Definition box). Stretch out the State effect over the whole timeline (or as desired).

[image]

I'm sure there MUST be an easier way to do that, but it's just what I came up with on the fly.

PaulNBerlin, 2020-01-04:
Hi, thanks a lot for your help.

I am struggeling with the model you send me.
What do I have to do make working it as a "tree"?

On the screenshot you can see that I inserted your model and a tree from xlights.
Adding the same effect to both result in the following:
1. Your tree changes all colours at ones (also changed the config a bit in other tries but nothing really worked as expected)
2. The xlight tree behaves as expected.

Is it possible sharing more that just the model?
[image]

[Video.zip](https://github.com/Aircoookie/WLED/files/4022606/Video.zip)

thedannymullen, 2020-01-05:
to make a tree use the megatree model.  See this link: https://youtu.be/2dJlWqKrsKg

If you have 16 50led strings tied to one d1 you need to set like i show in video.   Set as one string in xlights with 16 strands.  Then set nodes/string to 800.  In xlights a "string" is the number of controller connections a "strand" is the actual strings of leds in the tree.  It is very confusing the first time.

thedannymullen, 2020-01-05:
@samstreak to turn things off you use the off button in the sequencer.  
[image]

Look in upper left the black square.  

Turning things off with wled was an issue I had in xlights and switched back to espixelstick.  Wled defaulted back to the orange color when data not being sent.  So If you see orange you can try to set the default color to black.  This setting did not work for me when I tried.  I plan to debug further in a week when I take my light show down.

brentbrooks70, 2020-01-05:
@samstreak & @thedannymullen 
Wouldn't you want to click the icon second from the right (black lightbulb)? For me this turns my WLED connected LEDs off but does not stop transmitting to it so it does not revert back to WLED local effect. If I click the stop sign then it stops the E1.31 transmission and WLED reverts back to local control. I just started playing with Xlights about a month ago so I might be completely off base.

thedannymullen, 2020-01-05:
@brentbrooks70 Yes to quit sending the data to xlights you want to click the dark light by the stop sign.  

When wanting to turn lights off during a sequence you want to us the turn lights off.  For example I have a few effects that when the sequence is done the lights stay on.  I needed to go add the "off" effect so at the end of the sequence that light would turn off.  If I did not do this then going into the next sequence in my player the light stayed on.  

I misunderstood @samstreak question until I just re-read it.  He was wanting to turn off lights in the middle of a model.  When I first read I believed he wanted to turn off lights at the end of a sequence.  
He could actually do what he wants with the effect I sited I believe.  @samstreak I believe, if you created a group with the pixels you want to be off you could then use the "off" effect I stated on the group for the whole sequence.   Don't think I will have time to test this until next week.

PaulNBerlin, 2020-01-05:
Regarding Tree setting: I got it working to see the effects with both: Polyline from @samstreak and Tree.
Figured out that the polyline behaves like a tree when I add it to a group. Dont know why adding to a group is needed but I am happy that it is working.

[image]
[image]
[image]

thedannymullen, 2020-01-05:
Sounds good!!  Glad to hear!

samstreak, 2020-01-05:
I had to set the Render Style in the Layer Settings section to "Per Preview" to get my trees working right... sorry, forgot about that part.

[image]

samstreak, 2020-01-05:
So I found an "easier" way to get rid of the extra pixels in my tree model...  using submodels:

Go into the Tree1 model and down to the Sub Models at the bottom of the first section. Click the ... button.   Add a New Sub Model for each vertical section. I named mine Up1, Down1, Up2 (etc), and then on the right side specified the node numbers to include in that model:

[image]

After that, add a group for just the sub-models  (I named mine Tree1WithoutSkipPixels, but again, whatever makes sense to you), and make sure to check the "Show submodels to add" check box so you see the sub-models in the list of things to add to the group

[image]

Then you just use the new group in your sequences as you would any other model.  The render style still needs to be set to "Per Preview" for this to work right, but you can totally skip the State stuff I went through before.

Hope this helps!

PaulNBerlin, 2020-01-05:
@samstreak that is working fantastic. Thank you very much!
Also a big thank you to @thedannymullen

closed 2020-05-11 by stale[bot]