wled/WLED #358: not-to-sync lights are syncing anyway

prankousky, 2019-11-20:
Hi everybody,

I have `WLED` flashed to multiple `nodeMCU`s; details below... Even though those **should not** sync, they do. They have different mqtt topics, so I don't understand why they would sync at all.

I integrated those devices via separate `.yaml` files in Home Assistant; also, I believe they have automatically been discovered by Home Assistant as well..? When I change a device name (from `wled` to `wz_vorne`) in the webUI, it will update the name in Home Assistant.

There is an automation that starts the two wleds in the kitchen (controlled by `ku/all`, see below) when motion is detected there. However, when those devices turn on (and they successfully do when there is motion!), the `wz_vorne` wled will also turn on - **and even** another one (which has different mqtt topics than the kitchen and living room ones). 

I have turned off  `sync lights` for **all wled**s in the webUI like this

[image]

Below are configurations for two of those three devices; the third one is similar,as it does have a completely different mqtt topic (so there shouldn't be a reason for it to sync with the others).

The main issue here is that the third (not pasted) light is above the bed. If somebody triggers the kitchen motion sensor in the middle of the night, the wled in the bedroom will be turned on (alongside the ones in the kitchen that are meant to turn on as well as the one in the living room, which is *not* meant to turn on just like the bedroom one).

Would somebody please point me in the right direction on how to fix this? I have been looking through the `.ino` file, as I compiled the `.bin` file myself, but don't remember anything about hard-coding syncing all devices there...

Thank you for your help.

------

* "wz_vorne" (Living Room)

[image]

```
light:
  - platform: mqtt
    name: "WZ Vorne"
    command_topic: "wz/dc8f0a"
    brightness_command_topic: "wz/dc8f0a"
    rgb_command_topic: "wz/dc8f0a/col"
    rgb_command_template: "{{ '#%02x%02x%02x' | format(red, green, blue)}}"
    effect_command_topic : "wz/dc8f0a/api"
    effect_list:
    - "[FX=00] Solid"
    - "[FX=01] Blink"
    - "[FX=02] Breathe"
    - "[FX=03] Wipe"
    - "[FX=04] Wipe Random"
    - "[FX=05] Random Colors"
    - "[FX=06] Sweep"
    - "[FX=07] Dynamic"
    - "[FX=08] Colorloop"
    - "[FX=09] Rainbow"
    - "[FX=10] Scan"
    - "[FX=11] Dual Scan"
    - "[FX=12] Fade"
    - "[FX=13] Chase"
    - "[FX=14] Chase Rainbow"
    - "[FX=15] Running"
    - "[FX=16] Saw"
    - "[FX=17] Twinkle"
    - "[FX=18] Dissolve"
    - "[FX=19] Dissolve Rnd"
    - "[FX=20] Sparkle"
    - "[FX=21] Dark Sparkle"
    - "[FX=22] Sparkle+"
    - "[FX=23] Strobe"
    - "[FX=24] Strobe Rainbow"
    - "[FX=25] Mega Strobe"
    - "[FX=26] Blink Rainbow"
    - "[FX=27] Android"
    - "[FX=28] Chase"
    - "[FX=29] Chase Random"
    - "[FX=30] Chase Rainbow"
    - "[FX=31] Chase Flash"
    - "[FX=32] Chase Flash Rnd"
    - "[FX=33] Rainbow Runner"
    - "[FX=34] Colorful"
    - "[FX=35] Traffic Light"
    - "[FX=36] Sweep Random"
    - "[FX=37] Running 2"
    - "[FX=38] Red & Blue"
    - "[FX=39] Stream"
    - "[FX=40] Scanner"
    - "[FX=41] Lighthouse"
    - "[FX=42] Fireworks"
    - "[FX=43] Rain"
    - "[FX=44] Merry Christmas"
    - "[FX=45] Fire Flicker"
    - "[FX=46] Gradient"
    - "[FX=47] Loading"
    - "[FX=48] In Out"
    - "[FX=49] In In"
    - "[FX=50] Out Out"
    - "[FX=51] Out In"
    - "[FX=52] Circus"
    - "[FX=53] Halloween"
    - "[FX=54] Tri Chase"
    - "[FX=55] Tri Wipe"
    - "[FX=56] Tri Fade"
    - "[FX=57] Lightning"
    - "[FX=58] ICU"
    - "[FX=59] Multi Comet"
    - "[FX=60] Dual Scanner"
    - "[FX=61] Stream 2"
    - "[FX=62] Oscillate"
    - "[FX=63] Pride 2015"
    - "[FX=64] Juggle"
    - "[FX=65] Palette"
    - "[FX=66] Fire 2012"
    - "[FX=67] Colorwaves"
    - "[FX=68] BPM"
    - "[FX=69] Fill Noise"
    - "[FX=70] Noise 1"
    - "[FX=71] Noise 2"
    - "[FX=72] Noise 3"
    - "[FX=73] Noise 4"
    - "[FX=74] Colortwinkle"
    - "[FX=75] Lake"
    - "[FX=76] Meteor"
    - "[FX=77] Smooth Meteor"
    - "[FX=78] Railway"
    - "[FX=79] Ripple"
    - "[FX=80] Twinklefox"

```

* "ku_schrank" (Kitchen)

The `ku/all` topic is used by **two individual** WLED devices that are both located in the kitchen.

[image]

```
light:
  - platform: mqtt
    name: "Küche WLED All"
    command_topic: "ku/all"
    brightness_command_topic: "ku/all"
    rgb_command_topic: "ku/all/col"
    rgb_command_template: "{{ '#%02x%02x%02x' | format(red, green, blue)}}"
    effect_command_topic : "ku/all/api"
    effect_list:
    (... as above...)
```

Aircoookie, 2019-11-21:
Strange that this happens...
One thing to watch out for is to deactivate the "Send notifications on direct change" checkbox on every installation. Otherwise, the sync toggle in the web UI will turn on again and lights will sync after each reboot. It seems to be on for at least 1 kitchen device. If you are sure that you never want the lights to sync, the safest way is to set different UDP ports (just always add 1), that way sync can never happen.

On a side note, I noticed that your light has `ku/all` both as group and as device topic. That is not a good idea. You can use `ku/all` as device topic for one device (and leave its group topic empty) and as group topics for the other Kitchen devices.

I hope this will solve the issue!

Viele Grüße :)

prankousky, 2019-11-22:
Yay, vielen Dank! It seems to work now (by changing the UDP ports for each device). I did not yet change the topic, as it seems to not make a difference at the moment.

Question: I had previously created yaml files in Home Assistant for each WLED; this is not necessary any longer, as they are auto-discovered, correct? I deleted those yaml files and the WLEDs still worked (except some automations don't address their colors correctly and they turn white instead of colorX, but that might be due to the way I wrote those automations...)

Aircoookie, 2020-01-05:
Sorry for my late reply :)

Yes, that's correct, HA has a native integration since 0.102.0!

closed 2020-01-05 by Aircoookie