wled/WLED #360: Easy switch in the GUI for Normal, E131 and UDP

ricardogfp, 2019-11-20:
It would be nice to include a switch in the GUI to quickly switch between normal, E131 and UDP modes. For instance currently I have setup LEDFX as E131 and Hyperion with UDP, everytime I want to switch between dancing lights to TV ambilight or normal effect, I have to stop the streaming services. If there were a switch to stop listening to E131 and keep UDP or viceversa I could have the streaming services opened and just switch them by WLED GUI.

pana800, 2019-11-21:
I recently tried WLED and it is amazing. I was looking for this feature and found someone having the same interest. This is a nice feature and I would surely vote for it. Can have a switch or a trigger to overwrite the UDP by other effect if there is blank signal from Hyperion.
Thanks a lot

Aircoookie, 2019-11-21:
Although I would always recommend turning off Live streaming signal sources when not using them (as it adds load to your network and might impact the performance of your WLED installation), I will add such a toggle/feature to the upcoming new UI!

pana800, 2019-11-22:
Great thanks. Can it be an auto switch that can be triggered on by in coming UDP and off when there is no data?
Thanks again for such a great software.

eddiepropane, 2020-01-31:
Thanks for this and looking forward to it being implemented for my setup which is similar to above. 

A toggle will be great, at the moment I either disabling an instance of Hyperion or UDP in WLED. As suggested an automatic switch to normal mode on black in input from Hyperion would be awesome but not sure how feasible it would be to implement for you.

Araforn, 2020-02-17:
So I have been working on something very similar to integrate hyperion with diyhue.

See my post here. 

The issue with hyperion and a USB grabber is that it continues to send a UDP stream even on a no signal scenario.

I tried WLED today and of course its the same issue where Hyperion takes over 'Realtime' and doesn't give back 'Normal' control to WLED. 

Hence what led me to this thread. I have added an MQTT switch to my diyhue code and this blocks the UDP stream allowing diyhue control the strip again. 

Ideally the UDP stream should stop but this is not the case for most Hyperion users (that use a USB grabber) and requires a workaround. 

@Aircoookie do you plan on making the switch accessible by home assistant/MQTT. Really great work on WLED by the way!

Araforn, 2020-02-17:
> Great thanks. Can it be an auto switch that can be triggered on by in coming UDP and off when there is no data?
> Thanks again for such a great software.

An auto switch can only be designed if the signal is consistent. I have created an auto block of UDP on an always black signal (r=0,g=0,b=0) which is easy but if there is a continuous noisy signal (which is my case with hyperion "no signal" screen) then it becomes quite hard.

Even hyperion cannot manage to detect a no signal within its 'signal detect' settings.

Aircoookie, 2020-05-05:
This is now partially implemented in 0.10.0, you can override all live sources and run WLED effects. It doesn't support detecting and switching between different live sources though, but I cannot recommend running two live sources simultaneously anyways :) Will keep this open though

jes1417, 2020-05-08:
Is there will can call the switch manually with the API so we can use it in home assistant?

Araforn, 2020-05-08:
> Is there will can call the switch manually with the API so we can use it in home assistant?

This would be something I'm very interested in also as per my above comments either. Perhaps a way to toggle it over MQTT?

Aircoookie, 2020-05-08:
It is toggleable via the JSON api state `lor` property. See https://github.com/Aircoookie/WLED/wiki/JSON-API
Unfortunately no MQTT yet, but I gotta add JSON support to MQTT at some point soon!

blazoncek, 2022-02-21:
WLED supports simultaneous DDP and E1.31/ArtNet since 0.13-b3.
Also `/json/cfg` endpoint allows changing configuration _on the fly_ from client SW.
Closing.

closed 2022-02-21 by blazoncek