wled/WLED #361: Docs for compiling WLED for esp32 in PIO

plasmasolutions, 2019-11-20:
Hi @Aircoookie !

First of all: What a library! Judging from the sheer amount of useful features it's just everything that I ever dreamed of! 

As I was not able to get the binaries running on my esp32 here (flashed it with the esptool [is that even correct?], but it never created an softAP), I was attempting to compile your lib in PlatformIO, which was not successful either. Although I installed all libraries according to your quickstart, uncommented env_default = esp32dev in the platformio.ini and hit on Build, several errors were thrown at me, the first being:

> 
> Assembler messages:
> Fatal error: can't create .pio/build/esp32dev/lib744/NeoPixelBus_ID547/internal/NeoEsp8266UartMethod.cpp.o: No such file or directory
> *** [.pio/build/esp32dev/lib744/NeoPixelBus_ID547/internal/NeoEsp8266UartMethod.cpp.o] Error 1

Is there any documentation on howto build the library with OpenplatformIO or is it heavily experimental?

Thanks very much,
Tom

raoulteeuwen, 2019-11-21:
I read there are sometimes problems which can be solved by doing an erase of the flash (google "sudo esptool.py --port erase_flash").

You can also try another flasher tool: many people report good results using ESPhomeFlasher .

plasmasolutions, 2019-11-21:
Hi @raoulteeuwen ,

thanks for your heads up! Using esphomeflasher made it work! Looking at the code I see that he's erasing the flash before flashing - something that you mentioned as well, so that was most likely it!

Nevertheless I would love to be able to compile the library as I plan to contribute to it in the future. So any help on the compiling topic is much appreciated as well!

Greetings, Tom

closed 2019-11-21 by Aircoookie

plasmasolutions, 2019-11-22:
Hi @debsahu,

thanks for taking responsibility for my issue! It didn't fix it fully though - it just hangs at a later stage:

> Compiling .pio/build/esp32dev/src/src/dependencies/blynk/Blynk/BlynkDebug.cpp.o
> Assembler messages:
> Fatal error: can't create .pio/build/esp32dev/src/src/dependencies/async-mqtt-client/AsyncMqttClient/Packets/SubAckPacket.cpp.o: No such file or directory
> *** [.pio/build/esp32dev/src/src/dependencies/async-mqtt-client/AsyncMqttClient/Packets/SubAckPacket.cpp.o] Error 1

debsahu, 2019-11-22:
@plasmasolutions It compiles on TravisCI https://travis-ci.com/Aircoookie/WLED/builds/137580684 and also on my setup. Could you delete .pio .pioenvs and .piolibdeps directories, resave platformio.ini file, wait until PIO installs all required libraries and esp32 core (wait until intellisense index build is complete)
[image]
Then click compile and report back, and are you making any other modifications to the code?

plasmasolutions, 2019-11-23:
Hey @debsahu,
thanks so much for this idea! I deleted the .vscode and .pio folder - now it works perfectly! Deleting only .pio didn't do the trick.

Have a great weekend, 
Thomas

plasmasolutions, 2019-11-23:
Ok, last entry from me just to make sure to spread how I truly fixed it (in case anybody stumbles across this thread):

Even though I thought at first that deleting the folders fixed it I wasn't be able to build anything after the first successful build. Even resetting (fresh checkout) everything (including deletion of the folders) didn't work.

But one thing was strange: Intellisense was cyclicly rebuilding the index (even waiting for an hour didn't stop Intellisense from working) - so I disabled it in the "User Settings" in Visual Studio Code

It's this setting:
> "platformio-ide.autoRebuildAutocompleteIndex": false

It seems like Intellisense is locking files while analyzing them and that's what's causing the hickups. That explains as well why we don't see this behavior on TravisCI and other non-VSCode envs.

Cheers, 
Thomas

IronicBadger, 2019-11-26:
You are a legend!

> It's this setting:
>    "platformio-ide.autoRebuildAutocompleteIndex": false

Fixed it for me too.

enly1, 2020-05-25:
This is worth adding to FAQs. Spent many hours trying to compile this......

Aircoookie, 2020-05-25:
Added it to the FAQs: https://github.com/Aircoookie/WLED/wiki/FAQ#compilation-issues
Never personally had this problem, but seems like it is somewhat frequent :)

JWolfJr, 2020-06-12:
@plasmasolutions Thank you for posting your fix.  I also spent a lot of time on this, having the same issues.  Works perfect now! @Aircoookie Thank you for this app, it is truly awesome. Great work!

commit 095429a7 (2019-11-21): Merge pull request #364 from debsahu/patch-1

fix #361
commit e0283163 (2019-11-21): fix #361