wled/WLED #362: Problems when on static IP address

abarregar, 2019-11-21:
I would like to report that I have 3 NodeMcu's running your awesome WLED program. I have experimented with xlights and have been successful in running them on the E131 protocol as well. I have had rock solid performance on the NodeMcu's until I decided to put them on static IP's so I can use JSON api, or HTTP api requests in webcore. Once the NodeMcu reboots with the static IP it runs as per normal except about every 10 seconds it will pause for about 8 seconds. In any one of the FX Modes that has some movement the movement stops every 10 seconds for about 8 seconds and then restarts. If I put the NodeMcu back into DHCP it completely goes away.

A second thing of interest is that the sync mode has a mind of its own. I turnoff sync on either mobile app or browser page save and reboot and it turns itself back on. 

Anyways love your software and thank you and this great community you've built along with the many others.

raoulteeuwen, 2019-11-21:
Could you test with switching off NTP while using static IP and see whether that solves the freezes? A similar thing with NTP was mentioned in another issue...

Aircoookie, 2019-11-21:
I have almost all of my ESPs on static IP and haven't experienced issues... Have you set the gateway IP to your router IP?

abarregar, 2019-11-21:
Thank you both for the responses.

First yes the Gateway IP in the NodeMcu is set to the AP IP which is the DHCP in my case.

I used raoul's suggestion of turning off the NTP and had some success. 1 out of the 2 NodeMcu's I'm running immediately rectified itself and ran smoothly, and as you'd expect the clock was way out thinking it was 1970 again :-)  After some investigating on the second NodeMcu I noticed that there was a Blynk device auth token assigned and I reset that back to default. After I removed the Blynk token it is now running flawlessly on the static IP.

One thing I cant figure out is why the sync keeps turning itself back on. I deselect sync on the web browser interface save and reboot and it comes back.

Thanks for your help

Aircoookie, 2019-11-21:
Nice, good to know you worked it out!
Blynk and NTP are indeed the only remaining blocking APIs (aka they cause responsiveness issues if they fail to connect)

About the sync, if you want to have it off by default even after rebooting, go to sync settings and uncheck the "Send notifications on direct change" checkbox :)

closed 2020-04-05 by stale[bot]