wled/WLED #366: Lost connection

pepeEL, 2019-11-22:
Hi
I have write on nodeMCU v3 your firmware 0.8.6
I have setup fort this nodeMCU static IP but it is setup on my router. 
I also has added WLED in my Home Assistsant software and i see that from time to time devices in Home Assistant is not available and lost connection with WLED. nodeMCU has a good signal WIFI. I dont know where is a problem....? Please help me.

raoulteeuwen, 2019-11-22:
Hi. You say "nodeMCU has a good signal WIFI": how do you know/how have you measured (at the location the nodeMCU is now installed/do you have a measure at the same time Home Assistant reports a lost connection? Are you able to 'be at the LED strip when HA reports a lost connection', and check whether you CAN connect to WLED using a browser?

pepeEL, 2019-11-22:
I have nodemcu about router. When Home Assistant Lost connection i also can not connection by web browser

raoulteeuwen, 2019-11-22:
I assume "I have nodemcu about router" means "I have the nodemcu very close to the router"?

Can you try to disable NTP sync in WLED? There are users that report WLED 'freezing', and when they switch off NTP sync, that stops. I would be surprised if this is the problem in your case, but please try. Do you have another nodemcu that you can try, so we lower the chance it is a defective nodemcu? 

If the above doesn't help, you could try to reflash WLED. Most users report succes using ESPhomeFlasher.

And: what ESP does your nodeMCU contain, ESP8266 or ESP32? And exactly what file did you flash (exacte name of the bin from the github page)?

closed 2020-04-05 by stale[bot]