wled/WLED #369: Power supply, delay off

HillNet, 2019-11-22:
Hi,

Similar to #346, is there a way I can add a delay off on the relay toggle?

I'm using an ATX PSU and I'd like to keep the PSU running for, say, 30sec, after the LEDs are switched off.

FYI, I've compiled from source so I could change the relay mode from 1 to 0. 

Thanks for developing this software.

--
Roland

Aircoookie, 2019-11-23:
Hi!
Contrary to #346, this is really simple to achieve. Actually the software already has a pre-set delay for turning the relay off (0.6 seconds). You can change the delay in `wled09_button.ino`. Just change the 600 in [line 83](https://github.com/Aircoookie/WLED/blob/master/wled00/wled09_button.ino#L83) to, say, 30000 for 30 seconds until relay power off :)

HillNet, 2019-11-23:
Well that was embarrassingly easy...

Should have guessed you had that covered!

Thank you 😁

HillNet, 2019-11-23:
Solved

closed 2019-11-23 by HillNet