wled/WLED #370: WS2811 pixel nodes flickering

vamseekrishnaravi55, 2019-11-22:
Hello All - First of all, I appreciate all the time you guys are spending on this wonderful project which inspires many (like me) to do their DIY LED lighting. I'm completely new to this LED world and started to play with my Christmas lighting with WLED(0.8.6 version) and nodemcu. Everything looks good so far with all the effects on the LED strings except a blink(or flickering) with different color for every few seconds on the full sting(50 node string). I had very good connections and grounded properly between the LED string and nodemcu as well, but unable to fix the issue. Please give some recommendations to overcome this issue.

Thank you in advance.
Vamsee.

photocromax, 2019-11-23:
> Hello All - First of all, I appreciate all the time you guys are spending on this wonderful project which inspires many (like me) to do their DIY LED lighting. I'm completely new to this LED world [...]

Try to keep the data wire as short as possible, or use a thicker wire. 
Be sure the power supply and nodemcu share the same ground.

richyxi, 2019-11-23:
You can try putting skip the first led in the configuration for a boost in
the signal

coelner, 2019-11-23:
#230 maybe same as here?

palengh, 2019-12-03:
Thank you for your great effort with this project. I really appreciate it.
I too experience the random flashing of the lights.
Running WLED 0.8.5
My power supply is 12V (12.5 A) directly to the LEDs. My NodeMCU has D4 connected to the LED´s program wire. 
When setting Max current to 12500 mA, the LEDs flashes like strobe light. 
I need to set the LED Max current to below 2000 mA, to stabilize it, but even then it will flash randomly.
When I disconnect the D4 connection, it's stable, even at 12500 mA.
The D4 connecting cable is 8 cm long.
Setting "skip the first led" in the configuration dit not help

closed 2020-04-08 by stale[bot]