wled/WLED #373: No WiFi Connection after restart 0.8.6

r100gs, 2019-11-23:
Hello aircoookie,

Everything wirks fine in the beginning. I can Setup my network and it connects to my WiFi. But after unplugging it and repower again connection tobmy WiFi is lost and I have to flash and setup again.
Best regards,
Stefan

joshbob160, 2019-11-23:
I have this same issue. I cannot get it to connect to my wifi after setup. Settings seem to be saved for me though... I haven't needed to reflash and re-setup necessarily but it won't connect to the wifi unless it is plugged into my computer.

jefbenet, 2019-11-24:
I'm not sure if I'm having the same issue as others or not. I can connect to AP mode on initial boot, establish client/station parameters to connect to my home wifi, which is successful and I can interact as expected. But on powering down my esp32 by cable pull, or even by "save & reboot" command from 'Security & Update...' page, unit reverts back to AP mode every time, despite any attempt to change settings in 'Wifi Settings' page. Tried removing AP name to drop AP mode, changed setting from 'No Connection after boot' to 'Disconnected', and even tried 'Never'. Also my LED count drops to default of 30, despite changing to 32 and saving. In summary ALL settings are reverting to default on any type of power cycle. Using 0.8.6 "WLED_0.8.6_ESP32.bin" on Espressif ESP32-WROOM-32D. Any help would be greatly appreciated.

raoulteeuwen, 2019-11-24:
>  ... on Espressif ESP32-WROOM-32D. Any help would be greatly appreciated.

Could you check https://github.com/Aircoookie/WLED/issues/271 and see whether that brings you further?

r100gs, 2019-11-24:
OK, I have to wait around 2h then Connection is Back.
At least its working.

Best regards

Aircoookie, 2019-11-24:
@r100gs @joshbob160 thanks for reporting! It sounds like there is no issue with WiFi, but the settings can't be retained for some reason... It would be very helpful if you provide more info for me to determine under what circumstances this happens:
- If you save a preset in AP mode (the 1-16 slots you use to save favorite effects), will it still be loadable after a reboot or is it also gone?
- What is your exact ESP board?
- Did you compile from source or use the binary? If binary, what version did you try (filename) and which tool did you use?

@r100gs weird... if it doesn't connect within the first minute I wouldn't expect it to connect at all... although that is unlikely, maybe the signal is too weak for a reliable connection or the router has a DHCP lease/Connection limit. But good to know it now works! I hope you won't have to wait 2h for connection every time you reboot!

Def3nder, 2019-11-24:
Hi Aircoookie,

same for me with several ESP8266 / 8285 boards: the setting do not survive a power-off.
**None of the change survive - it's not only the WiFi settings.**

Another strange thing I noticed (perhaps it has something to do with it):

when you enable RGBW in the code (bool useRGBW = true;) then the white LEDS aren't working until you go into LED settings and press SAVE without changing anything.

**When I can test something, please let me know.**
I do have a lot of boards to use: ESP01, WeMos D1 mini, WeMos D1 mini pro, NodeMCU, some 8285 borads, ...

r100gs, 2019-11-24:
Hello Aircoookie,

I use WLED_0.8.6_ESP8266.bin (nothing self compiled)  on WEMOS D1 mini with SK6812.
I use 
`esptool.py --port /dev/ttyUSB0 erase_flash`
`esptool.py --port /dev/ttyUSB0 write_flash -fm dout 0x00000 WLED_0.8.6_ESP8266.bin`
to flash on linux manjaro

After some trying I think Wifi is working, but the webserver does not respond. 
The settings survive a repower cycle, but are slow and choppy in the beginning after a restart.
It takes some time to stabilize than everything works fine.

Physikus74, 2019-11-24:
Hello,
Parts of the described problem look familiar to me:
I had downloaded the finished binary file for ESP32 and flashed it into my board. (AZDelivery ESP32 NodeMCU Module WLAN WiFi Development Board ) After that WLED always answered with its own access point. No mather what I set about it, WLED did not log on to my WIFI. I also thought that was due to the WIFI configuration. When trying to control the light I noticed that no setting was available. Finally, I compacted the sources myself. After that everything worked fine, even the old settings were available again. Maybe this will help you with your troubleshooting...

Translated with www.DeepL.com/Translator

Def3nder, 2019-11-24:
I just did a recompile with same sources but changed to Arduino IDE instead of PlatformIO.
And then the settings in WLED survived a power cycle.

I found out that the problem was that I used a 4M falsh version for the ESP8285 (that has only 1M flash). 
I asume that the EEPROM will begin @ the end of the flash and this would be Flash RAM that isn't there.

so for me the following lines in **platformio.ini** did do the job:
```
[env:esp8285]
board = esp8285
platform = espressif8266@1.8.0
monitor_speed = ${common.monitor_speed}
upload_speed = ${common.upload_speed}
framework = ${common.framework}
build_flags =
  -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH
  -Wl,-Teagle.flash.1m0.ld  ;;;; Compile with no SPIFFS to leave space for OTA
  -D WLED_DISABLE_BLYNK
  -D WLED_DISABLE_CRONIXIE
  -D WLED_DISABLE_HUESYNC
  -D WLED_DEBUG
lib_deps =
  ${common.lib_deps_external}
```

faljse, 2019-12-05:
After soldering my quinled board (ESP32) i re-tried using WLED yesterday and yet again settings are not saved.
Also presets seem to dont work at all.

I flashed current version via platformio and also ota-flashed the pre-build pin16/esp32 version.
Same issue with both builds.

Flashion via esptool for ESP32 never worked for me (boot fails) - are there instructions somewhere on how to do this?

Def3nder, 2019-12-05:
Hi,

It seems you have chosen a different flash size for compiling them you ESP
has. Did you check that the settings of the board in the platformio.ini
match to the board you are using ?

Amli2011, 2019-12-07:
I am new to the WLED firmware, and prefer precompiled binaries. 
I did try the diferent versions for esp01 version 8.6.
Did also try it on diferent known good ESP01-modules. It did always end in a bootloop.
After reading i tried the wled_0.8.6_ESP8266_lwip1.bin. and was very happy as it did not bootloop. But then I ran into exactly the problems described here:
Could connect to the AP and set credentials for my network. When I hit save it connected to my network and everything was working until i pull the power (or reboted under system settings) Then all settings was lost and I was back to start - It booted to AP-mode.

The reason was obvious (after å lot of thinkering.....) - the lwip1 version is compiled for 4MB, and the ESP01 modules I have is only 1M. When I realised that I tried with an ESP12 module witch works flawless.
Just in case anyone got the same problem and tried the same solution - that led to this problem...

faljse, 2019-12-11:
> I flashed current version via platformio and also ota-flashed the pre-build pin16/esp32 version.
> Same issue with both builds.

I re-tried with arduino; which did work and now i noticed it also does work with platformio.
I modified platform/board for d1mini32, but it seems to be unnecessary, so it could have beeen because of:
- erase_flash
- some commit to WLED
- stupid me

JeroenSchwab, 2020-02-01:
Hello,
I had the same problem saving on a nodeMCU v3 (nonofficail) to make it work i had to use:
esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB -fm dout 0x00000 WLED_0.8.6_ESP8266.bin

with the 4MB instead of 1MB

Def3nder, 2020-02-01:
Hi @JeroenSchwab, yes, this is correct. A NodeMCU has 4MB (or 16 Mbit) flash - why did you think to use the 1M version ?

JeroenSchwab, 2020-02-02:
Hi @Def3nder, I don't know why but I thougt i'tt mention it for other people like me.

Def3nder, 2020-02-02:
HI @JeroenSchwab, yes, you are totally right - **sorry for my disturbing reply.** 
There are quite a few pitfalls to step into and "just flashing an binary" is one of them.

Thanks for highlighting this: the binary needs to be build for the same flash size as the target device.
"Intuitively" I would think that as smaller bin would fit into a bigger flash, but on the ESPs this does not work.

kombi21251, 2020-04-04:
Hello! I have the same problem.
Changing the settings of the router helped me.
I am changed the channel of wireless network from "auto" to "1".
My router is TP-LINK AC750.
I am use WLED_0.8.6_ESP8266.bin

But not for long. After some time, WLED disconnected: / I select another channel in the WLED settings, and set the automatic channel on the router. WLED connect. But I do not see the WLED IP device on the router. Despite this, WLED is available on 192.168.0.100

closed 2020-08-09 by stale[bot]

Craft-im, 2023-03-18:
I think i found solution... my led was connected to gpio2, as default and wemos not connected to wifi after reboot board. Then i connect data pin to gpio4 (D2 on board) all works fine)))

blazoncek, 2023-03-18:
GPIO4 is **not recommended**.
Use GPIO2 or GPIO1. If ESP fails to boot add 4k7 Ohm pull-up resistor (to 3.3V).