wled/WLED #374: Would like some help compiling

pierslawson, 2019-11-23:
Hi

I am also having problems with E1.31 as described in #355 so wanted to compile the latest master to see if it helped. I'm using:

Arduino IDE 1.89
ESP8266 2.5.2
NeoPixelBus 2.5.1
FastLED 3.3.0
ESPAsyncWebServer 1.2.3
ESPAsyncTCP 1.2.2
ESPAsyncUDP  1.1.0
IRRemoteESP8266 2.7.0

I'm compiling for the "NodeMCU 1.0 (ESP-12E Module)"

But I get the following errors about multiple definitions:

```
c:/users/pierslawson/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\ESP_Async_E1.31\ESPAsyncE131.cpp.o:(.rodata._ZN12ESPAsyncE1316ACN_IDE+0x0): multiple definition of `ESPAsyncE131::ACN_ID'; sketch\src\dependencies\e131\ESPAsyncE131.cpp.o:(.rodata._ZN12ESPAsyncE1316ACN_IDE+0x0): first defined here

c:/users/pierslawson/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\ESP_Async_E1.31\ESPAsyncE131.cpp.o: in function `ESPAsyncE131::parsePacket(AsyncUDPPacket)':

C:\Users\pierslawson\Documents\Arduino\libraries\ESP_Async_E1.31/ESPAsyncE131.cpp:102: multiple definition of `ESPAsyncE131::parsePacket(AsyncUDPPacket)'; sketch\src\dependencies\e131\ESPAsyncE131.cpp.o:sketch\src\dependencies\e131/ESPAsyncE131.cpp:97: first defined here

c:/users/pierslawson/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\ESP_Async_E1.31\ESPAsyncE131.cpp.o: in function `ESPAsyncE131::initUnicast()':

C:\Users\pierslawson\Documents\Arduino\libraries\ESP_Async_E1.31/ESPAsyncE131.cpp:57: multiple definition of `ESPAsyncE131::initUnicast()'; sketch\src\dependencies\e131\ESPAsyncE131.cpp.o:sketch\src\dependencies\e131/ESPAsyncE131.cpp:54: first defined here

c:/users/pierslawson/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\ESP_Async_E1.31\ESPAsyncE131.cpp.o: in function `ESPAsyncE131::initMulticast(unsigned short, unsigned char)':

C:\Users\pierslawson\Documents\Arduino\libraries\ESP_Async_E1.31/ESPAsyncE131.cpp:69: multiple definition of `ESPAsyncE131::initMulticast(unsigned short, unsigned char)'; sketch\src\dependencies\e131\ESPAsyncE131.cpp.o:sketch\src\dependencies\e131/ESPAsyncE131.cpp:65: first defined here

c:/users/pierslawson/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: libraries\ESP_Async_E1.31\ESPAsyncE131.cpp.o: in function `ESPAsyncE131::begin(e131_listen_t, unsigned short, unsigned char)':

C:\Users\pierslawson\Documents\Arduino\libraries\ESP_Async_E1.31/ESPAsyncE131.cpp:40: multiple definition of `ESPAsyncE131::begin(e131_listen_t, unsigned short, unsigned char)'; sketch\src\dependencies\e131\ESPAsyncE131.cpp.o:sketch\src\dependencies\e131/ESPAsyncE131.cpp:37: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```
I've not used the Arduino IDE much before... so perhaps I'm just making a newbie error... or the wrong version of a library?

Any help appreciated... otherwise, when might you drop another release with the E1.31 changes in place?

Estyn, 2019-11-24:
Not at all an expert, just started playing around with this today, but if you open it in Visual Studio Code it will install the platform.io extension and resolve all the dependencies for you.

pierslawson, 2019-11-24:
Ahh... I hadn't come across PlatformIO before. I was using the Arduino plugin. It now seems to build and I can upload to the board. Once uploaded I can connect to the web site hosted by the board.

However, after resetting the board (or powering cycling) the web site becomes unavailable. I see the light on the board flasing off then back on again every few seconds like it is crashing and restarting. Using the Monitor in PlatformIO I see:

```
Ada

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

I can't expect much as the code from master is not necessarily tested... but any thoughts on whether I'm doing something daft? Steps so far:

1. In PlatformIO use env:nodemcuv2/Build
2. In PlatformIO use env:nodemcuv2/EraseFlash
3. In PlatformIO use env:nodemcuv2/Upload
4. Connect to the WLED AP
5. Browse to 4.3.2.1 
6. Setup WiFi
7. Connect back to WiFi
8. Browse to WLED UI
9. Edit LED settings
10. Restart board
11. Browse to WLED UI - no response

Thanks!

pierslawson, 2019-11-24:
I've just tried:

1. In PlatformIO use Clean at the top level
2. In PlatformIO use Update Project Libraries
3. In PlatformIO use env:nodemcuv2/Build
4. In PlatformIO use env:nodemcuv2/Upload

And it seems to be working fine!

Thank you for the pointer to PlatformIO. Now back to looking at E1.31 problems!

closed 2019-11-24 by pierslawson