wled/WLED #375: WiFI slow to response when leds are on

c-soft, 2019-11-24:
Hi, thanks for great work! 

The issue I'm having is as follows: after successful setup, everything seems to be working great, except from sluggish UI. But sluggish UI seems to be only slow when leds are on. The issue seems to be caused by slow ping times - see image below. Initial pings are with leds off, then I turn on leds (no animations!) and ping increases. Unfortunately this gets worse over time to the point that UI is completely unusable till next reboot :(

[image]

Aircoookie, 2019-11-24:
Hi, thanks for reporting!

That's not good, seems like the ESP is somehow overloaded, but this shouldn't happen, much less so without an effect (solid mode). Could you send me the output of `192.168.2.218/json/info` before and after you turn on the LEDs?

c-soft, 2019-11-24:
Of course, this is in OFF state:

```
{"ver":"0.8.6","vid":1910255,"leds":{"count":345,"rgbw":true,"pin":[2],"pwr":445,"maxpwr":20000,"maxseg":10},"name":"WLED","udpport":21324,"live":false,"fxcount":83,"palcount":50,"arch":"esp8266","core":"2_5_2","freeheap":17216,"uptime":17369,"opt":119,"brand":"WLED","product":"DIY light","btype":"bin","mac":"840d8ead5293"}
```
... and below is in ON state:

```
{"ver":"0.8.6","vid":1910255,"leds":{"count":345,"rgbw":true,"pin":[2],"pwr":3384,"maxpwr":20000,"maxseg":10},"name":"WLED","udpport":21324,"live":false,"fxcount":83,"palcount":50,"arch":"esp8266","core":"2_5_2","freeheap":16608,"uptime":17450,"opt":119,"brand":"WLED","product":"DIY light","btype":"bin","mac":"840d8ead5293"}
```

c-soft, 2020-01-02:
I have at the moment 4 instances of DYI lights done on WLED, only one of them has this communication issue. I'm quite confident this has to do with my hardware setup there.
I havent quite noticed slugishness of the UI after time, I don't think it's the software issue after all. Closing.

closed 2020-01-02 by c-soft

mahdemir, 2025-05-22:
> I have at the moment 4 instances of DYI lights done on WLED, only one of them has this communication issue. I'm quite confident this has to do with my hardware setup there.
> I havent quite noticed slugishness of the UI after time, I don't think it's the software issue after all. Closing. 

Did you solve the issue, i have the same problem?