wled/WLED #376: MQTT-Support Port 1883

McDepp, 2019-11-24:
I have a strange behaviour when I connect my Wemos D1 mini R1 running to an MQTT-Server listening to a port other than 1883 ( in my case 1884):
A soon as WLED connects to the MQTT-Server it freezes totally. The only thing to get through this issue is to flash the Wemos again and I have to set the Arduino IDE to ERASE FLASH: All Contents.
I tried this behaviour with versions 0.8.5 und 0.8.7, three different Wmeos and a NodeMCU E12. Its everytime the same issue.
I even can't post any debug log files as there aren't any to post. I see the typical start messages and that's all. Sorry.

Aircoookie, 2019-11-25:
Thanks for the report! That indeed doesn't sound good.

@ everyone: does someone have WLED connected to a broker on a different port? I do not know why using port 1884 could even have the potential to cause issues

McDepp, 2019-11-25:
Hi Aircookie,
thanks for Feedback. 
I have to run the MQTT-Server on Port 1884 because I use IoBroker, own a lot of Sonoffs amd the Sonoff-Adapter of IoBroker runs on Port 1883. WLED connects to the Sonoff-Adapter but the adapter seems to expect certain topics.

brentbrooks70, 2019-11-26:
@Aircoookie I have 2 NodeMCU, 2 D1 Mini, 1 Adafruit feather esp8266s and 1 Adafruit huzzah feather 32 and 1 generic esp32  boards all running MQTT on port 21883 with no detectable issues at all. I will get the versions when I get home and post but there should be nothing below the 0.8.6 pre-release but I’ll double check.

rdlaner, 2020-01-03:
@brentbrooks70 you mentioned successfully using WLED on the Huzzah32 feather board. Did you have to make any pin config changes or did you just use the esp32.bin file? I'm looking at the Adafruit schematic and it doesn't appear that GPIO 2 is made available on this board. I'm hoping you can straighten me out. Thanks!

brentbrooks70, 2020-01-04:
@rdlaner I had to compile with my own pin defined. I have used GPIO**26** (A0), GPIO**25** (A1) and GPIO**4** (A5) in the past with success, currently I am using GPIO4 on both of my Feather32s running WLED. 

And to verify my previous statement from 11/26/19...All boards running 0.8.6 or above (actually now they are all running 0.9.0-b1 commit bd4d1cd or above)

rdlaner, 2020-01-04:
@brentbrooks70  Awesome, thanks for the info!

brentbrooks70, 2020-01-05:
@rdlaner FYI, the reason I used those pins is because they are not usable as analog pins while WiFi or BT radios are operable  as they are on ADC2

closed 2020-05-11 by stale[bot]

djcyph, 2020-05-26:
> @rdlaner I had to compile with my own pin defined. I have used GPIO**26** (A0), GPIO**25** (A1) and GPIO**4** (A5) in the past with success, currently I am using GPIO4 on both of my Feather32s [...]

I know this old but any chance you could send me a copy of your binary for the huzzah?  I'm still learning how to compile with arduiono and haven't quite got there yet to make my own.  thanks!