wled/WLED #377: Nodemcu not stable

phdelodder, 2019-11-24:
I have flashed wled 0.8.6 on a nodemcu with 16 led. I'm using it in combination with Home Assistant. 

I can't get a stable connection, using either dhcp or static ip. 

Any Ideas?

raoulteeuwen, 2019-11-24:
- is the nodeMCU responding when you connect to it in your browser / on IP-basis?
- is the nodeMCU responding when it is still in AP mode?
- how far away from your router/AP is the nodeMCU? Could it be a 'normal' connection problem?
- do you have another ESP based device laying around you can try?

(fyi: i have 2 nodeMCU's running WLED 0.8.6 fine ... most of the time ... 1 is sometimes off line, but that is also when other devices in it's vicinity are having problems, so for sure a connection problem)

phdelodder, 2019-11-25:
I have now placed the nodemcu with issues next to a nodemcu without issues and it remains te same. According to the wifi log on the router the the connection is stable, although according to HA it's highly unpredictable the connection. 

It is responding IP based or AP, the duration is is the problem. I'll see if I can find another board and replace it. I have had issues with tasmota in the past with arduino core version.

raoulteeuwen, 2019-11-25:
So i understand that you have no problem connecting to the nodeMCU from a mobile device.

Feels like it might be HA and it's polling that might have a 'different tolerance' on reporting the nodeMCU as on- or offline? Feels like a 'false positive', HA reporting 'falsely' that the nodeMCU is offline?

What info / log etc from HA is it you use exactly, and what is the exact error HA produces?

phdelodder, 2019-11-25:
That's true, I have had a connection test running randomly and no issues what so ever. 

Will see of if on HA there is room for improvement: https://github.com/home-assistant/home-assistant/issues/29043

closed 2019-11-25 by phdelodder