wled/WLED #379: http polling

bwente, 2019-11-24:
I see that there are option from controlling WLED from outside your local network. I was wondering if it would be possible to add a simple polling method? Rather than exposing the ip externally, have the ESP check a config/settings file every 15 minutes or so. It could be a static JSON or XML file or dynamically generated. You could use dropbox to sync and update all of them at once.

Aircoookie, 2019-11-25:
Hi! While that would certainly be possible, I don't believe it would be very useful to many users. The perfect protocol for cross network control is [MQTT](https://github.com/Aircoookie/WLED/wiki/MQTT) though! You can use a public broker like _broker.hivemq.com_ and when you push a message to the device topic it will instantly get set to the ESP, no polling required.

bwente, 2019-11-25:
I didn't know there was public broker. I haven't used MQTT, so that's something new for me. I was trying to do a simple cheerlight, for family members.

closed 2020-04-05 by stale[bot]

nothans, 2022-10-11:
> 

CheerLights now spports MQTT natively and with Adafruit IO and ThingSpeak. (mqtt.cheerlights.com:1883 / https://cheerlights.com/cheerlights-now-supports-mqtt/)