wled/WLED #380: random flashes and corruption

JamesCordell, 2019-11-24:
Hi 

Great project and I appreciate the hard work go in. I do have an issue however,

I have a strip of 555 lights. with good parallel power supplies at regular intervals. I am using an esp32 and I have noticed the first 214 led flash and appear to flash randomly. Only in specific modes and operations. When in lighthouse mode. If I drop the FX speed below half the glitch appears with random flashing on the first 214 lights. If its above it works correctly without flashing.  Another interesting mode is pallet. The first 214 leds flash incorrectly. 

I have discovered the 214 is relative to my 555 lights. I have narrowed the corruption down to if I only use 350 leds it works perfectly if I then go to 351 the first light flickers and corrupts. This is only in certain modes. Pallet always has the corruption.

Aircoookie, 2019-11-25:
Hi! Thanks for the report!
I will need to try that out, seems like the driver struggles to push more than 350 LEDs/or a certain framerate.

JamesCordell, 2019-11-26:
This issue has since disappeared in version 0.8.6! only prevalent in 0.8.5.
Thanks

closed 2019-11-26 by JamesCordell