wled/WLED #382: WLED restarts if being controlled via E1.31 with more than ~256 LEDs and accessing the web interface

lonestriker, 2019-11-24:
First really have to say WLED is incredible, thanks so much for the software!

I'm using 3 NodeMCU boards with WLED to drive WS2812B strips with either xLights or Falcon Player (FPP).  This situation causes the controller to restart:

* If I control more than ~256 LEDs (haven't found the exact number, but it's > 256 and < 300)
* Playing a sequence via FPP or xLights over E1.31 unicast
* Then try accessing the web interface

If I use the WLED built-in effects, driving up to 450 lights (most I've tried) does not cause any issues.  It's only when using E1.31 and trying to access the web interface that the controller restarts.  I have three different NodeMCU boards controlling different numbers of LEDs. One controls 249, one controls 100, the last one controls up to 450 lights.  The 249 and 100-LED ones never have any issues with restarts.  The 450 one does randomly while playing a sequence (likely random things pinging the web interface).  It's not a power issue as I inject power every 150 LEDs; and if I just reduce the number of lights controlled, things work flawlessly.

I suspect the controller is just too under-powered to play a sequence with too many lights and try to provide the web interface at the same time?  Is there a way to lock out the web interface while it's running a sequence to prevent restarts?

Thanks again!

Aircoookie, 2019-11-25:
Hi! I'm glad you like WLED!
I am sorry that you are experiencing crash issues...
What WLED version are you running?
I just switched to E1.31 async in source and will publish a release soon! With that, the issue should be fixed (webinterface is still accessible even with over 500 LEDs in my testing)

lonestriker, 2019-11-25:
0.8.6 latest release at the time.

Thanks so much for the quick reply and the pending fix!

ggrant111, 2019-11-27:
I had that issue when using Unicast. I switched to multicast and it hasn’t reset on that. Using 8.5 and 8.6. 

Side question how to I control more than one universe on a single controller. I setup in xlights starting universe 1 and three universes. I have 360 leds connected to one wled controller. But when I play the sequencer it only lights up to 170 of the pixels. Which I assume is just the first universe.

lonestriker, 2019-11-27:
I had your issue with multicast and only controlling a single universe :)  I was never able to get past that problem so I just went back to unicast and then ran into the resetting issue.

FYI, as a test, I loaded ESPixelStick firmware on my device and it had no issues keeping up or resetting.  I hadn't noticed previously but the 0.8.6 version of WLED also lagged slightly when pushing data from Falcon Player to the longer 256+ LED string.  But, ESPixlStick firmware lacks a lot of features WLED has; though it's really good at primarily E1.31 interaction with the LEDs.  I couldn't get multicast to work with ESPixelStick at all.

ggrant111, 2019-11-27:
Ok I’ll try using Unicast and see if it will ouput more than 1 universe.

When I used it before the controller would reset the WiFi settings. And I
would have to do a hard reset, unplug and plug back in to get it to reset.

I was using jinx during that time and not xlights. Not sure if that would
make a difference.

lonestriker, 2019-11-27:
You can try flashing ESPixelStick while you wait for WLED 0.9.0 to be released to fix the reset issue.  You might have more luck with using it for multicast than I did.  The flashing program jar and the firmware can be downloaded here:
https://github.com/forkineye/ESPixelStick/releases/tag/v3.0

ggrant111, 2019-11-27:
Thank you. Just changed to Unicast and used xlights. Now it’s showing all
three universes on the strips as expected. Must be an issue with multicast
it needed to change the settings in wled to see more than the first
universe.

Three hours trying different settings. Thanks for your input.

starwolf73, 2019-12-03:
Similair issue here. I setup a panel of 32x8 WS2812B leds, a total of 256 leds using 2 universe on a Wemos D1 mini with WLED 0.8.6 firmware binary and using Jinx! to test it. It was working perfectly nice and very smooth. I add a second panel of 32x8 and extend the universe 2 and was still working perfectly. 

Soon I try to add universe 3 for the rest of the second panel for a total of 512 leds, the portion of the universe 3 leds was lagging/hanging and even totally wrong but universe 1 and 2 was still working fine. Unfortunately the Wemos D1 mini start been unstable and reboot every 15-20sec.

Test:
- External labs power supply 5v 5A with WLED current limited at 4000ma.
- I have double check both 32x8 panels standalone and work fine with WLED effect and 2 universes.
- Using WLED buildin effects with both 32x8 panels work fine, no glitch or lagging.
- I test on different Wemos D1 mini and same results.

I know ESPixelStick binary is also running the ESP8266 CPU at 160MHz, is the WLED running at 80Mhz or 160Mhz?

lonestriker, 2019-12-05:
Just tried the [0.9.0 beta build](https://cdn.discordapp.com/attachments/473448917543944193/651956100555603988/wled_1912043_ESP8266.bin) and it works perfectly. No more lagging effects and hung web interface for long strings > 250 LEDs!  Thanks!

starwolf73, 2019-12-05:
@Aircoookie The 0.9.0 beta build is definitely more stable and doesn't reboot with 512 leds but I still experience hiccup and small hang which doing text scrolling or some effect doesn't look good. I test on both software Jinx! and Xlights and still have the same effect on both. 

Will it be possible to compile the same version but with CPU at 160Mhz like the ESPixelStick? This is from [ESPixelStick](https://github.com/forkineye/ESPixelStick) (For best performance, set the CPU frequency to 160MHz (Tools->CPU Frequency). You may experience lag and other issues if running at 80MHz.)

PS: Do you have donation or Patreon link? I definitely like version 0.9.0 interface and would like to provide small compensation.

Mariu86, 2019-12-06:
You can support Aircoookie here https://paypal.me/aircoookie

ggrant111, 2019-12-06:
Is there a new version for the ESP32? Or just the 8266 for now?

starwolf73, 2019-12-06:
> @Aircoookie The 0.9.0 beta build is definitely more stable and doesn't reboot with 512 leds but I still experience hiccup and small hang which doing text scrolling or some effect doesn't look good. [...]

@Aircoookie Thanks for your hard work and keep up the work. A little compensation been send to you ;)

illuxions, 2019-12-11:
@Aircoookie I really appreciate your great toys and help!

I have been content with ESP32(wroom) I had to slow my show down to 10fps and occasional stack dumps but I thought it was just something inherent.  Multicast or Unicast they work ok.

0.8.6 was ok, and most of the recent commits have been getting better and better but on a whim I tried flashing the latest commit to one of my NodeMCU 12e  that I was initially having problems with on 0.8.6.  WOW,  What a difference, no lagging at all and runs like a hose.  I flashed them side by side wiping the whole memory before flashing then put used xlights test to demonstrate the difference. 

The ESP32 is jerky and clearly dropping some serious frames.  I tried disabling everything I could before build, it didn't seem to help much. I tried a hex inverting schmitt trigger to buffer the levels up to 5v and it didn't help. I tried board_build.f_cpu = 240000000L board_build.f_flash = 80000000L under env:esp32dev and a few variants to see if it was hitching internally.  I'm not sure what is going on.

Here's a quick vid with the latest commit built and uploaded in VS Code. (it's unlisted) https://youtu.be/2wDSAMu0oXg 

Have you seen any odd things with the ESP32?  It has a lot more horsepower than the 12e, so something is screwy.  

For testing they're on the same lone subnet with the beaglebone with Falcon player and 8 string cape running as a bridge between my main net and test subnet with only those 3 things on that entire wifi/router(Asus RT66u).  The buffer I tested is a Ti 74HC14.  Power supply is a 20A 5V meanwell set to 5.1v when everything is off, sagging to 5.0v with all 300 pixels all white 50%.

I just ordered 6 more NodeMCU 12es to replace the ESP32s in my display.  I'd like to use the ESP32s but as smooth as the 12es are I'm hooked. I'll sneak you some more beer money when my wife isn't sitting behind me.

closed 2019-12-13 by Aircoookie