wled/WLED #383: No access to NodeMCU after 0.8.6

TheZoker, 2019-11-25:
I have a NodeMCU with a cp2102 and after upgrading from 0.8.5 to 0.8.6 the chip restarts all the time (all LEDs go on and off every ~10 seconds) and I have no access to the device anymore. When I flash 0.8.5 again, the board works again without any issues. 

Is there anything I provide to debug the issue?

chunkysteveo, 2019-11-27:
Yeah, your mention is correct behaviour for my ESP-01 1Mb - after WiFi config and reboot, it appears (the newer ESP-01S boards have a power LED and it flickers off/on every 10 seconds or so) to reboot, and the LED's attached to it go out, and then back on, like on boot up. Back to 0.8.5 and all is fine.

Aircoookie, 2019-11-27:
@TheZoker I believe it was a mistake on my side to "trust" lwip 2 as the default option for WLED 0.8.6, it had bootloop issues before, but they appeared to be fixed for me... Did you already try the binary with the lwip1 suffix?

@chunkysteveo yes, I believe the bootloop on ESP01 is the same issue (mDNS library and lwip2 not playing well together somehow). But there is one additional problem. Lwip 1 sketch size is bigger, so it would no longer fit on ESP01 512k. 1m will still work, but no OTA updates will be possible.

By the way, 0.9.0 will be the last version that doesn't require SPIFFS, which will take up at least 32kB. So starting from 0.10.0 I unfortunately need to drop 512k support (1m will work nicely, but will be cumbersome, because you'd need to update manually via the flasher every time).

chunkysteveo, 2019-11-27:
Thanks for the info @Aircoookie , good to know! I don't think it's unreasonable to expect such a huge project to fit into a 512k board anymore - there are a lot of other solutions for addressable LEDs that work well if space is an issue. 

I'm using ESP-01(s) boards on some ESPixel Pop boards that I made up, purely to make them as small as possible. I think even a Wemos is a little overkill for driving just one chain of LED at Christmas for me (and maybe others). I have even looked into the relatively simple procedure of replacing the flash chip with a 32Mb version to get the ESP-01 up to 4MB, and it's about $2 and my time to replace all my ESP-01 chips - lots of fun!

Keep up the good work @Aircoookie !

TheZoker, 2019-12-13:
@Aircoookie Is this issue resolved with 0.9.0?

Aircoookie, 2019-12-13:
It should be since I am using lwip 1 for compilation again. But feel free to try it out, if it works nicely, we can close this :)

chunkysteveo, 2019-12-13:
Just uploaded to 0.9.0-b1 to my ESP-01 (1Mb) module (just one for testing) via the Manual OTA Update option, using release bin file "WLED_0.9.0-b1_ESP8266_1M_ota.bin" and it worked (even did it remotely via VPN from work!!). Still testing etc but it's not stuck in a boot loop at least - so I think it should be fixed? :)

Steve

TheZoker, 2019-12-16:
Seems to work for me as well! Thanks!
I'm going to close this issue

closed 2019-12-16 by TheZoker