wled/WLED #384: Nodemcu and led flashing randamly

art353, 2019-11-26:
I am using a nodemcu controller and CS2803 (Upgraded WS2812B) led  with a 5v 60amp power supply  i am controlling 608 led in series with power injection every 6 meters and agen at the end everything was working fine at 426leds
So my issue when i connected the last segment it was set to solid red but what i see is red and green then it starts to blink and randomly change color at different speeds (agen its supposed solid) i discounted the last segment but it continues rebooted to factory settings and continued to flash but if i tuch the nodmcu it gose solid i figerd the controller is shot but what could have caused this

brentbrooks70, 2019-11-26:
@art353 If touching it makes a difference I would double check all of your connections and grounds. Does it fix the issue with 608 when you touch it or just the 426? If you do not inject power at the end, what is the voltage reading at end of 608? 426? If you connect that last 182 LED segment directly to the controller (or another one) without the previous 426 connected does it work as expected? Could you also please post the results provided from http://_DeviceIP_/json/info?

art353, 2019-12-13:
You were completely right sorry it took me so long 
So originally I wireed 22 gauge wire from the nodemcu  to the power supply. D4 was only 6 inches from the nodemcu everything was soldered together so i added a  nother ground wire from the nodemcu to the led 6 inches away and that completely resolved the problem

brentbrooks70, 2020-01-05:
@art353 Is there a reason you reopened this? Do you need further assistance? If not would mind closing? Thanks

closed 2020-01-05 by art353