wled/WLED #386: ESP-01 1Mb and Release 0.8.6 Binary stuck in boot loop

chunkysteveo, 2019-11-26:
I would say this is more of an FYI to an issue that is already noted in the release notes of 0.8.6 - but I am uploading the binaries for several ESP-01 1Mb (some with Puya flash chips, some without) and using the WLED_0.8.6_ESP01_1m_full.bin and WLED_0.8.6_ESP01_1m_ota.bin uploaded using PyFlasher resulted in the ESP-01 either getting stuck in a boot loop after configuring WiFi credentials, or it would always lose the Wifi Details on each reboot and not save them.

I reverted to the 0.8.5 release WLED_0.8.5_ESP01_1m.bin and it is working perfectly fine. Would love to see a new release which incorporates all of 0.8.6, and boots and saves correctly. Happy for now just to get some Christmas lights working using a bunch of ESP-01 modules!

Keep up the great work, happy to test further if needed - not tried compiling my own firmware yet.

TheZoker, 2019-11-26:
I guess your issue is related to my issue: #383

chunkysteveo, 2019-11-27:
Yeah, your mention is correct behaviour for my ESP-01 1Mb - after WiFi config and reboot, it appears (the newer ESP-01S boards have a power LED and it flickers off/on every 10 seconds or so) to reboot, and the LED's attached to it go out, and then back on, like on boot up. Back to 0.8.5 and all is fine.

chunkysteveo, 2019-11-27:
Going to close this one @Aircoookie , please read [issue 383](https://github.com/Aircoookie/WLED/issues/383) for updates and issue with lwip.

closed 2019-11-27 by chunkysteveo