wled/WLED #387: Analog Clock Customization

Sasquatch6ft9, 2019-11-26:
More Settings for Analog Clock Overlays,  Colors and Brightness, Maybe a "off" pixel before and after the hour, minute, second and 5 minute "markers/Hands" to make them stand out with an effect still running behind them

I'd really like a clock only with the above and 24 hour options etc 

Your Analog is the easiest/best I've found so far

sjude68, 2019-11-27:
Yes, marking the hands standout would be good. It hides behind the effect.

knochenkopp, 2019-11-29:
And maybe "overlaping colors" like https://github.com/leonvandenbeukel/Round-LED-Clock
He mixes the colors, so when the red one hits the blue one it turns mangenta or if the green one is in the same position as the blue one it lits up in cyan.

Aircoookie, 2020-03-29:
Will add an option for turning off the pixels next to the "hands" soon!

user-51, 2020-06-08:
Feature request: Add an option to turn off the clock overlay as soon as an effect (i.e. anything besides "all pixels black") is started.

snijhuis, 2020-09-26:
I do like the idea of being able to set the clock brightness, preferably from Home Assistsnt. I'm probably going to use WLED in a dedicated clock and it would be great if I can dim the clock during evening of during movie watching for example.

cshotton, 2021-01-28:
> Will add an option for turning off the pixels next to the "hands" soon!

An option for RGB values for the hand colors would be great, too, as long as we're jamming more stuff into the settings. ;)

nickizgr8, 2021-07-19:
I've been playing with the WLED analog clock quite a bit lately, using different sizes and materials (epoxy, wood, metal, etc.) to make really cool wall and desk clocks, I love this clock feature. I would like to upvote the comment by @cshotton , being able to change the colors of the minute, second, and 5-min marks without having to compile binary would be amazing!

sansillusion, 2021-07-20:
I also think it would be great. I made one a while ago I called an eclipse
clock.

https://youtu.be/It_TK4mL4nk

BlackCatPeanut, 2022-12-27:
Just replaced a Tasmota clock firmware with WLED. This does everything I need bar the colour changes to the "hands" as described in comments above. Would love to see those as options. In the meantime I'll look into the compiling option as a workaround

blazoncek, 2023-01-25:
There were recent enhancements in 0.14 to analog clock overlay as well as a few usermods dealing with analog clocks.

closed 2023-01-25 by blazoncek

markabyrd, 2024-04-20:
Closed as "completed" ?  Seems the status is Closed as Ignored.  No 0.14 enhancements address this issue.  
Who authored the Analog Clock Usermod?  Can they be encouraged (or compensated) to implement the customization request?

markabyrd, 2024-05-25:
My upvote on any additional customization options for Analog Clock.  Don't know where to start to try and add function to WLED through my own analog clock mod or add new function to a current usermod.  Does WLED have a community that develops usermods for hire?

blazoncek, 2024-05-25:
Use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

softhack007, 2024-05-25:
The analog clock usermod allows customization of colors, maybe try it instead if the buildin clock feature:

https://github.com/Aircoookie/WLED/tree/0_15/usermods/Analog_Clock

build_flags `-D USERMOD_ANALOG_CLOCK`