wled/WLED #389: [proposal] Add a collection of useful usermod configuration

mrVanboy, 2019-11-26:
Hi,
thanks for an awesome project and great job!

I would like to propose an idea of creating a collection with useful `usermod` configurations. I believe a lot of users solving similar problems like adding displays, additional input/outputs, etc. 

What do you think about having a separate folder like `examples/usermod` or something similar (maybe wiki with a list of snippets) where we can contribute our existing pieces of code?

Aircoookie, 2019-11-26:
Hi,
actually I've been thinking about this as well! I was going to create a separate repository, but a folder is indeed a better idea since it is easier to find and everyone who adds an usermod will be listed as contributor.

Thanks for the suggestion, just created the directory :)

mrVanboy, 2019-11-28:
Thanks for the fast response! I'll prepare the first PR during the weekend probably. Closing the issue as it's solved I believe.

closed 2019-11-28 by mrVanboy