wled/WLED #392: Start effect/animation from the start when the lights are turned on.

ikrowni, 2019-11-27:
I think i had an issue opened for something else and mentioned this before but was hoping it was or could be fixed now. I have a strip of lights going up my stairs and a motion detector at the bottom and at the top. When motion is triggered at the top of the steps rather than starting the animation over from the top it starts the animation from wherever it was when it was last stopped.

Aircoookie, 2019-11-27:
Hi, sorry for not getting back to you on this!
I have yet to come up with a way to make starting effects from the beginning again without breaking timebase syncing (two ESPs synced together will always run the effect at the same progress).
But i'll see what I can do!

huggy-d1, 2019-12-11:
IF SYNC_IN = SYNC_OUT = _DISABLED_ THEN
  TRIGGER_ANIMATION_FROM_START = TRUE
ELSE
  TRIGGER_ANIMATION_FROM_START = FALSE
END IF

Then when a trigger's leading edge occurs:
IF TRIGGER_ANIMATION_FROM_START THEN
  RESET_CURRENT_ANIMATION(Animation(AnimationSelection))
END IF

I understand it is not as simple as that.  Perhaps could be with a couple helper functions.

I know it cannot be that simple, but you get the idea.  Otherwise, I think it would be too complicated.

KROWN, would you mind sharing how you accomplished this stairway lighting project using WLED?  I want to do the exact same thing.  I have all the hardware, but have no idea how to setup the Arduino environment to build a custom firmware .bin file -- or were you able to manage that using the stock default .bin?

ikrowni, 2019-12-11:
Yea I'm just using the stock release. Right now it works properly about 75% of the time until I have the ability to actually stop and start the lights where I want them I'm stuck with it like this. Hoping aircookie saves the day with the next release. But if you find something before then please let me know.

huggy-d1, 2019-12-11:
How did you get the motion sensor input working with the stock build using a quinled dig uno?

ikrowni, 2019-12-11:
oh im just using my motion sensors through (home assistant) to trigger the lights.

closed 2019-12-13 by Aircoookie

nibblerrick, 2020-05-31:
Is there now a way to trigger an animation from start? I haven't found something about that in the wiki - or I don't see it.

Aircoookie, 2020-05-31:
@nibblerrick there is no way to trigger this explicitely, but most effects (e.g. Wipe) will now start from the beginning when LEDs are turned on!

nibblerrick, 2020-05-31:
Ahh, thanks very much! Need to play around with an new LED-strip, I think... :-)