wled/WLED #405: Strange behavior after wifi connection times out

miraixyz, 2019-12-01:
Hello

I just programmed my nodemcu board with the 0.8.6 wled firmware.  I am using a static IP configuration, and it seems to drop pings from time to time which is no big deal.

However, when I turn off the LEDs for the evening, it will power back up after a while.  I notice this happens when the Nodemcu board loses connection to the wifi signal.

Another thing I noticed is that if I have wled cycling through my presets, it will stop cycling and go back to the default preset if the wifi loses connectivity. 

I suspect that that the time out period is too short or that there is no timeout programmed into the firmware.  Or maybe the program restarts if it detects signal loss?  Can this be fixed? Perhaps a longer timeout?

thanks

miraixyz, 2019-12-01:
Figured it out....  

I unchecked the "Turn LEDs on after power up/reset" option in the LED settings menu.  Fixed everything!

thanks

Aircoookie, 2019-12-03:
Sorry for my late response and good to know that you fixed the most annoying part of the issue!
The behavior you describe means that the board self-resets/crashes from time to time. I need to find out why that happens :)

closed 2020-04-08 by stale[bot]