wled/WLED #407: Which binary for Wemos D1 Mini?

SteveRMann, 2019-12-01:
Which binary should I download for the Wemos D1 Mini?

Aircoookie, 2019-12-03:
I'd recommend the `wled_0.8.6_ESP8266_lwip1.bin`! Should be the most stable.

r100gs, 2019-12-04:
Hello,

I tried it this morning (very short) but after disconnecting it from computer and reconnecting to an usb-charger works great.

closed 2020-01-05 by Aircoookie

robyevolution, 2020-02-27:
hello, where can i find the : wled_0.9.1_ESP8266_lwip1.bin  ??

Aircoookie, 2020-02-27:
@robyevolution this file doesn't exist, all 0.9.1 binaries are compiled with lwip1 :)
You can just use the regular `wled_0.9.1_ESP8266.bin`!

robyevolution, 2020-02-27:
I had the version  wled_0.8.6_ESP8266_lwip1.bin, I tried to install via OTA the WLED_0.9.1_ESP8266_1M_ota.bin but not work, why?

austwhite, 2020-02-27:
@robyevolution what device are you installing on?
Have you tried one of the other binaries?

Might be best to ask this as a new question as this one is closed and it is a different question to the topic of this issue.

robyevolution, 2020-02-27:
i have a wemos d1 mini

fishbone-git, 2020-02-27:
What aspect didn't work?  Did the OTA fail, and did you get any error messages? 
If the OTA succeeded (I'll assume it also then worked well enough to go back to the previous binary).  What wasn't working?

As above, probably best to take this elsewhere as there are now better options for handling questions like this: https://github.com/Aircoookie/WLED/issues/658

Aircoookie, 2020-02-27:
@robyevolution use the normal `WLED_0.9.1_ESP8266.bin` please, the 1M is only for devices with 1mb of flash like ESP-01 or Sonoff.
You are not the first one to choose the incorrect binary, I really have to make a page that explains which binary to use for what board :)

austwhite, 2020-02-27:
@Aircoookie 
Maybe need less binaries 🙂

Aircoookie, 2020-02-27:
Of course, there will be less once LED pin and type are configurable in the web UI 😄