wled/WLED #409: Running lights in parallel from a single NodeMCU

tika911, 2019-12-02:
Thanks for a great LED lights system.  
Not sure if this is an issue but I have run out of ideas.

I am attempting to run 4 Christmas tree decorations, each has 180 5v WS82811 LED's. using a 20A power supply.  If I connect a single string of 180 lights, everything works great.  If I connect more that one string in parallel all the lights turn white and are not controllable.  If I attempt to change colors the lights will flicker but remain white.  I figured I needed to use a load level shifter to boost the d4 signal to the light string.  I get the same result using a LLS.  If I only connect a single string to the LLS it works, more than one string same issue.

Can you use WLED to run more multiple lights in parallel?  Or is running lights in series only supported?

Mariu86, 2019-12-02:
Can you make a picture of your setup ? If everything it's wired correctly the light should work !

Aircoookie, 2019-12-02:
Hi, the 3.3v voltage is probably not sufficient for driving more than one string. You can wire them in parallel by connecting a single WS2811 to the ESP and than connect the output of that LED to your four inputs. Then you can also check "Skip first LED" in LED settings to disable the single LED.
I hope it will work :)

robkleinpeter, 2019-12-02:
Wouldn't all 4 power supplies need to share a common ground?  I defer to the experts but would think they should for the data to be sent/received correctly.

tika911, 2019-12-03:
I am only using a single power supply and everything shares a common ground.

Here is a pic of the way it is wired.  This is not my setup but a pic of the setup I used to get it working with a load level shifter.
[image]

Coming off the LLS is 5v going to the led's.  This works.  Now when I want to connect an additional string in parallel, I would assume I connect an additional wire to D4 to the right side of the LLS and the second string data wire to the corresponding left side pin on the LLS.  This does not work.  All the lights turn white and are not controllable.

Same thing happens if I use a sacrificial pixel in place of a LLS.  Single string works, more than one same white lights. This is leading me to believe that the issue is not a voltage drop.

robkleinpeter, 2019-12-03:
...using that pic, you should only need to connect all strips data to Left-Row 9.   No Right side changes needed.  If that doesn't work, refer to @Aircoookie response.

Aircoookie, 2019-12-03:
Yes, what @robkleinpeter said! Just connect all strips to one output of the LLS in parallel, leave the input as is.

tika911, 2019-12-03:
Forgot to mention that I tried that already.  If I connect more than one string to the left side of the LLS (row 9) all the lights are white and not controllable.

Then I went to splitting the d4 on the right side of the LLS shifting and connecting each string to a separate output but....same result.

robkleinpeter, 2019-12-03:
Humm, that seems like it should work but again I'm by no means an expert.  I did read [this](https://www.reddit.com/r/arduino/comments/3lyqdn/adafruit_500_ohm_resistor_on_data_channel_why/) a while back about putting a resistor between the data pin and LED to help with signal reflections.  Seems logical this could be an issue especially with 4 stips/strings on one data pin.  Sorry I couldn't be more help.

Mariu86, 2019-12-03:
With 2 strips should work but I think its something we missing, all the strips are the same 5v  ? The data signal it's in correct direction?

tika911, 2019-12-03:
I figured it out.  Bad ground on the wires that I was using.  I completely re-wired everything with new cable going to the LED's and it works like a champ!  Thanks everyone for the help

closed 2019-12-03 by tika911

robkleinpeter, 2019-12-03:
Awesome!  Glad it was something pretty easy to fix.  Welcome!

Mariu86, 2019-12-03:
It's working with 4 strips and level shifter? You tried to see if it's working without the level shifter ? It's good to know for future reference.

tika911, 2019-12-03:
I will check and let you know once I get the new wire in from amazon

tika911, 2019-12-08:
It works without the load level shifter

Thanks,

Greg