wled/WLED #410: WLED HOME FLASHER DOES NOT OPEN

nelson2305, 2019-12-03:
HI I am trying to download this app for Mac but once it downloads and I click open no action or window pops up. HELP!

Aircoookie, 2019-12-03:
Hi, do you mean this tool? https://github.com/esphome/esphome-flasher
I unfortunetely do not own a mac or use that tool for flashing. I'd recommend using esptool, it is command-line, but very simple: https://github.com/espressif/esptool

swaxx-ha, 2019-12-06:
Hi there, i use the esphome-flasher all the time. It may be the problem that you need to navigate around the fact that the software is unsigned. Therefore, MacOS will not run it per default. Please refer to this[ Suppport Page ](https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac)to learn how. Hope that it solves the problem. Cheers

tpmodding, 2020-01-14:
https://github.com/marcelstoer/nodemcu-pyflasher 

is use this on my mac, works very good!

closed 2020-05-20 by stale[bot]